vhelin/wla-dx #260: Blocks cannot have spaces in their names

maxim-zhao, 2019-09-29:
````
.block "Hello world"
nop
.endb
````
Gives:
````
foo.asm: INTERNAL_PASS_1: Unknown internal symbol "w" in ".wla1664a" at offset 97687.
````
I guess the linker format is using spaces as separators.

maxim-zhao, 2019-09-29:
This is at commit 98cefb145df5ff1c65388e4ee8b172106a9bb468.

vhelin, 2019-09-29:
WLA's internal format was yeah using spaces as separators... Now we encode space in the .BLOCK name with a ´, which seems to work. :) A quick hack, yeah, but works as no-one is using ´ in their block names (I hope)...

I hope this is enough...

maxim-zhao, 2019-09-30:
Better to use a non printing character in my opinion, to minimise the chance of someone typing it.

Best might be to use ASCII 31 (Unit Separator) instead of space as the delimiter, as it's the intended use and unlikely to be typed, but that's a wider ranging change.

vhelin, 2019-10-01:
Ok, instead of using the block's name directly in the WLA's internal data stream I just place a block name ID there, which is an integer, and this now works with all the block names WLA can parse.

maxim-zhao, 2019-10-08:
Works great!

closed 2019-10-08 by maxim-zhao