vhelin/wla-dx #262: Assembler spins forever

maxim-zhao, 2019-10-12:
On https://github.com/maxim-zhao/psrp/commit/2a1d28d10fe0a5f58dc6a7ba74637c9afa2f6dec WLA-Z80 consumes 100% CPU forever with no console output. I learned to use `git bisect` and got this:
````
5951f3f1245da9ca351082b4ed5fb26e8a88ea70 is the first bad commit
commit 5951f3f1245da9ca351082b4ed5fb26e8a88ea70
Author: Ville Helin <ville.helin@gmail.com>
Date:   Tue Oct 1 20:58:22 2019 +0300

    Added a better solution to GitHub issue #260.

 CHANGELOG |  2 +-
 defines.h |  6 ++++++
 main.c    | 11 ++++++++++-
 pass_1.c  | 21 +++++++++++++--------
 pass_3.c  | 32 ++++++++++++++++----------------
 5 files changed, 46 insertions(+), 26 deletions(-)
````
So I guess I was premature saying that #260 is good :) Unfortunately my repro case is enormous, but I guess it's related to use of `.block` (with or without spaces in the name).

vhelin, 2019-10-12:
Sorry about that, I should not code when I'm really tired. :)

maxim-zhao, 2019-10-12:
I agree - https://xkcd.com/323/ is the one to aim for :)

closed 2019-10-12 by maxim-zhao