vhelin/wla-dx #263: (Proposal) Variable-width string mapping

maxim-zhao, 2019-10-23:
This is a sort of extension to the `.asciitable` functionality, where instead of mapping single characters to single bytes, we instead map variable numbers of characters to variable numbers of bytes. The translation would match the longest possible "key" at the current point in the input string, and then emit whatever the output data is for that.

There are a few motivations:

1. Variable length strings are a good substitute for UTF-8 support - ANSI C code can see "オ" in UTF-8 as three or so chars and we can then map that to a single byte translation.
2. It could be useful for placeholders, e.g. I can map ASCII to 0-96 and then have "control sequences" like `<item>` map to single-byte codes above that range - nicer than having to inject `.db`s in the middle of my `.asc`s
3. We might actually want variable-length translations too
4. Sometimes I want 16-bit outputs for my `.asciitable` and the filter-macro version is really ugly

I can see that it might be painful to implement the "range" feature of `.asciitable` here because byte sequences from UTF-8 might well not come out as intended - although sequential Unicode codepoints will have mostly-sequential trailing bytes, there will be jumps every 64 code points. However the long-hand version is not the end of the world (people already make .tbl files this way).

...speaking of which, maybe this can be supported by reading a .tbl file? This tends to be in the form `<hex>=<text>`, where `<hex>` can be multiple bytes and `<text>` is some sort of text which may evaluate to multiple non-control-character bytes.

maxim-zhao, 2019-10-25:
Possible example syntax...
````
.dstringmap script ; named map because I may need more than one
; UTF-8 means the values are multi-byte
$00="ァ"
$01="ア"
$02="ィ"
$03="イ"
; ...
; Control codes
$ff01="<number>"
$ff02="<player>"
$ff03="\n" ; Doesn't matter if this is interpreted as one or two chars
.endd

.stringmap script "アイ<player>\n"
````
I'd suggest making it an error if you encounter any unmappable characters in the string.

It's a bit overwhelming to consider adding this myself... are there any docs around that might guide me towards things like how to add directives and the collection types WLA DX already has?

vhelin, 2019-10-29:
Sounds like an interesting feature!

I don't think we have any guides or documentation around about adding new directives to WLA DX. The starting point for a directive is in pass_1.c. Search for e.g., "directive_asctable_asciitable" in there for reference.

maxim-zhao, 2020-12-04:
Thanks for accepting my PR 😃 I'll be making use of this in an upcoming disassembly to support Japanese text in resources. I may drop a PR if anything else comes up (I noticed one or two things that could be tidied up too).

maxim-zhao, 2020-12-05:
Well, it's all working just as I wanted. Time to go back to a few projects and undo some of the hacks...

closed 2020-12-05 by maxim-zhao