vhelin/wla-dx #265: Clarify how to make install to directory other than /usr/local

pinobatch, 2019-11-27:
Under Ubuntu 18.04, how do I tell WLA-DX to install somewhere other than `/usr/local`? The `INSTALL` file didn't specify which variable to set, so I read `cmake_install.cmake` for more clues. But my best guess failed.

Doing this:
```
cd ~/develop
git clone https://github.com/vhelin/wla-dx.git
cd wla-dx
export PREFIX="$HOME/.local"
export CMAKE_INSTALL_PREFIX="$HOME/.local"
cmake -G "Unix Makefiles" .
make -j4
make install
```
Produces this error message on the final command:
```
-- Install configuration: "RelWithDebInfo"
-- Installing: /usr/local/bin/wla-gb
CMake Error at cmake_install.cmake:47 (file):
  file INSTALL cannot copy file "/home/pino/develop/wla-dx/binaries/wla-gb"
  to "/usr/local/bin/wla-gb".

Makefile:139: recipe for target 'install' failed
make: *** [install] Error 1
```

pinobatch, 2019-11-27:
After a bit more investigation of how to pass variables into CMake, this worked:
```
cmake -D CMAKE_INSTALL_PREFIX="$HOME/.local" -P cmake_install.cmake
```
Should it be documented this way? Or should the `install` target of the generated `Makefile` be changed to somehow pass `PREFIX` from the environment?

vhelin, 2019-11-27:
I have to admit that I know almost nothing about CMake. :) Help, anyone?

Neui, 2019-11-27:
AFAIK the `PREFIX` are make variables and would usually come after the make command. I thought CMake supported that, but apparantly, it doesn't, just `DESTDIR` (which isn't `prefix`).

Anyway, yes setting `-D CMAKE_INSTALL_PREFIX` at either configuring time (when calling `cmake .` in your example) or at install time (like you did, before the `-P`).

I think I had a branch where I had somewhat more informative README, let me see...

vhelin, 2022-03-12:
It seems that INSTALL.md talks about -D CMAKE_INSTALL_PREFIX so I guess this issue can be closed...

closed 2022-03-12 by vhelin