vhelin/wla-dx #267: wla-65816 likes generating 8-bit opcodes for immediate operations, even in 16-bit mode

SolraBizna, 2019-11-28:
At some point since 9.9, `wla-65816` has lost the ability to automatically emit immediate operands of the proper width.

Example source file:

```65816
.MEMORYMAP
DEFAULTSLOT 0
SLOTSIZE $40
SLOT 0 $0000
.ENDME

.ROMBANKSIZE $40
.ROMBANKS 1

.ORG $00
	; 8-bit accumulator
	SEP #$20
	.DB 'A'
	ADC #$69   ; right opcode
	.DB 'B'
	ADC.B #$69 ; right opcode
	.DB 'W'
	ADC.W #$69 ; manually generate wrong opcode
	.DB 'E'
.ORG $10
	; 16-bit accumulator
	REP #$20
	.DB 'A'
	ADC #$69   ; bug! generates 8-bit opcode
	.DB 'B'
	ADC.B #$69 ; manually generate wrong opcode
	.DB 'W'
	ADC.W #$69 ; right opcode
	.DB 'E'
.ORG $20
	; 8-bit index
	SEP #$10
	.DB 'A'
	LDX #$A2   ; right opcode
	.DB 'B'
	LDX.B #$A2 ; right opcode
	.DB 'W'
	LDX.W #$A2 ; manually generate wrong opcode
	.DB 'E'
.ORG $30
	; 16-bit index
	REP #$10
	.DB 'A'
	LDX #$A2   ; bug! generates 8-bit opcode
	.DB 'B'
	LDX.B #$A2 ; manually generate wrong opcode
	.DB 'W'
	LDX.W #$A2 ; right opcode
	.DB 'E'
```

Hex dump when assembled with 9.9:

```
00000000  e2 20 41 69 69 42 69 69  57 69 69 00 45 00 00 00  |. AiiBiiWii.E...|
00000010  c2 20 41 69 69 00 42 69  69 57 69 69 00 45 00 00  |. Aii.BiiWii.E..|
00000020  e2 10 41 a2 a2 42 a2 a2  57 a2 a2 00 45 00 00 00  |..A..B..W...E...|
00000030  c2 10 41 a2 a2 00 42 a2  a2 57 a2 a2 00 45 00 00  |..A...B..W...E..|
00000040
```

`00` and `20` are the 8-bit versions, `10` and `30` are the 16-bit versions. `A` precedes the Automatically-determined length, `B` the Byte length, `W` the Word length, and `E` follows the code. Note that `00` and `20` have a two-byte instruction following `A` and `10` and `30` have a three-byte instruction.

Hex dump when compiled with the most recent git:

```
00000000  e2 20 41 69 69 42 69 69  57 69 69 00 45 00 00 00  |. AiiBiiWii.E...|
00000010  c2 20 41 69 69 42 69 69  57 69 69 00 45 00 00 00  |. AiiBiiWii.E...|
00000020  e2 10 41 a2 a2 42 a2 a2  57 a2 a2 00 45 00 00 00  |..A..B..W...E...|
00000030  c2 10 41 a2 a2 42 a2 a2  57 a2 a2 00 45 00 00 00  |..A..B..W...E...|
00000040
```

Notice that it no longer matters whether the processor is known to be in 8- or 16-bit mode. `A` is now always followed by a two-byte instruction.

(`.ACCU`/`.INDEX` etc. have the same effect as `REP`/`SEP` according to my testing.)

KungFuFurby, 2019-12-02:
I traced the commit that caused the bug to 5b3ad22ceae9ae1d6c7c5201d814fdd8a537336b .

vhelin, 2019-12-03:
Thank you guys for the good report! I'll try to fix this issue...

vhelin, 2019-12-03:
There was a bug in parse.c (the #ifdef has incorrect define) which meant that the following few lines were never used in WLA-65816. I removed that ifdef trying to unify all the platforms operand hint wise, so now the code was used, and the code itself is really something that I don't think will work, and it doesn't as it generated the problems you are now facing...

#if defined(W65186)
    if (d > 0xFFFF && d <= 0xFFFFFF)
      operand_hint = HINT_24BIT;
    else if (d > 0xFF)
      operand_hint = HINT_16BIT;
    else
      operand_hint = HINT_8BIT;

vhelin, 2019-12-03:
Another way to solve this is to change decode_65816.c:

Original:

    /* select the mode */
    if (operand_hint == HINT_8BIT)
      y = 0;
    else if (operand_hint == HINT_16BIT)
      y = 1;
    else if (accu_size == 8)
      y = 0;
    else
      y = 1;

New:

    /* select the mode */
    if (accu_size == 8)
      y = 0;
    else if (accu_size == 16)
      y = 1;
    else if (operand_hint == HINT_8BIT)
      y = 0;
    else if (operand_hint == HINT_16BIT)
      y = 1;

I'll have to think which solution is better...

SolraBizna, 2019-12-03:
> @SolraBizna Can I copy your test code and use it as a test for WLA-65816 in the examples dir?

Sure, if it's suitable.

vhelin, 2019-12-03:
The first solution (remove code) caused some problems on other target CPUs so I think I'll use the second solution (prioritize accu_size and index_size). I'll commit and push the solution to this later this evening...

vhelin, 2019-12-03:
How about now with the latest sources? I hope it's better now...

SolraBizna, 2019-12-03:
As of 328d246 the bug seems fixed. Thanks!

closed 2019-12-03 by SolraBizna