vhelin/wla-dx #269: Object header flag checks should be more liberal

maxim-zhao, 2019-12-06:
When linking multiple object files together, the linker requires that some flag bytes (and the empty fill byte) are the same between object files. This means for example that you have to define `.smsheader` in every file if you want only one - but some source files may be cross platform, so it seems a shame to require that. 

Thus it seems to me some of the flags should be “tri-state” - where set, they should match, but if unset it doesn’t matter. Others should probably be only once - you can’t define the header twice.

pinobatch, 2019-12-06:
The [forum post that prompted this issue](https://www.smspower.org/forums/17742-WLADXMultiFileProjects) had this zipfile as its attachment, which we may be able to use as a test case:
[different-projects.zip](https://github.com/vhelin/wla-dx/files/3933359/different-projects.zip)

vhelin, 2019-12-09:
Sounds useful! I'll see what I can do later this week I hope...

vhelin, 2019-12-11:
Ok, I made the header bit checks more liberal. It should be now even possible to combine e.g. a SPC700 and 65816 object files. How does it work for you guys?

pinobatch, 2019-12-13:
It's working for me now. Thanks. Now I can continue with breaking my first SMS tech demo into multiple source files along the same lines as the NES and Super NES demos it was ported from.

closed 2019-12-15 by vhelin