vhelin/wla-dx #270: 65816 operand hints being ignored

aexoden, 2019-12-06:
After testing the newest version, it does seem to be having trouble for me if applying the .w to the actual operand:

```
LDX #$0100.w
```
The above doesn't work, as I get an "INPUT_NUMBER: Out of 8-bit range." error, but moving the .w to LDX (i.e. LDX.w) does work.

vhelin, 2019-12-07:
In the latest sources it's so that .INDEX/.ACCU/SEP/REP define the size of the arguments for mnemonics that use accumulator / index and operand hint is ignored. In what kind of a context you are using LDX?

I'm just trying to figure out should I somehow still make operand hinting to override .INDEX/.ACCU/SEP/REP...

aexoden, 2019-12-07:
It's a 16-bit index register context, but there's no real way wla-dx could know that. The program effectively does a single REP #10 on initialization and never touches it again.

vhelin, 2019-12-07:
Ok, I made it so that if the user manually makes an operand hint, then it comes first. Internally generated operand hints are ignored in this case. If there is no manually given operand hint then SEP/REP/.ACCU/.INDEX information is used. This should be the same functionality as before...

aexoden, 2019-12-07:
Hmm, even with that update, it still seems the case that LDX.w #$0001 will work, but LDX #$0001.w won't. Still an error about being out of 8-bit range.

vhelin, 2019-12-07:
If you take a look at examples/65816/opcode_test/main.s, in the last 'block' (05) we have 8-bit index using SEP #$10, but "LDX #$A2.w" which will override that 8-bitness of SEP... Are you sure you have the latest sources?

vhelin, 2019-12-07:
I added "LDX #$0100.w" to my test as well and it works...

aexoden, 2019-12-07:
I'm going to try to have to figure out a test case, since your simple example does indeed work, but my larger program is failing. How confusing.

aexoden, 2019-12-07:
Actually, maybe the difference is that I'm not actually using real immediate values, but labels that are defined in a ramsection in another file.

vhelin, 2019-12-07:
If you can create a small test case that demonstrates this problem, that would be really helpful! And if we can add it to the examples directory as well then we can prevent it from breaking in the future as well. :)

aexoden, 2019-12-07:
Sorry, I was wrong again, it wasn't a ramsection variable. It appears to be either values I've defined or potentially other cases. Here's an example that fails:

```
.memorymap
        slotsize $8000
        defaultslot 1
        slot 0 $0000
        slot 1 $8000
.endme

.lorom

.rombanksize $8000
.rombanks 1

.define test_var1 $0600

.struct test_struct
        test_entry dsb 1000
.endst

main:
        ;LDX #test_var1.w
        LDX #_sizeof_test_struct
```
Either of those two instructions will give the error for me. (Can only test one at a time, since it'll stop parsing at the first failure.

vhelin, 2019-12-07:
Thanks, I'll check out what's wrong...

vhelin, 2019-12-07:
I found a bug in my code, I was still deducing the operand hint in one case even though the user code has explicitly given the operand hint. If you now test with the latest sources, does it work?

aexoden, 2019-12-07:
It looks like it's working now. Thanks for dealing with my terrible bug report. (It started as a comment on a closed issue, but I decided to make it a new issue instead without actually checking what I had written.)

I'll close this for now.

closed 2019-12-07 by aexoden

vhelin, 2019-12-07:
No worries! Thank you for your report. Without it the problem would still exist. :)