vhelin/wla-dx #271: Z80: Allow 1- and 2-operand forms of ALU instructions with A

pinobatch, 2019-12-09:
I've hopped platforms from the Game Boy to the Game Gear, which involved switching from RGBDS to WLA DX. I've noticed that among assemblers for 8080-family processors, wla-z80 is stricter in some ways about syntax than RGBASM is.

In RGBASM, the ALU block ($80-$BF, $C6, $CE, $D6, ..., $FE) allows specifying A as the destination operand or leaving it implicit. Thus it treats these as synonyms:

- `add a, b` and `add b` -- WLA accepts the former
- `adc a, b` and `adc b` -- WLA accepts the former
- `sub a, b` and `sub b` -- WLA accepts the latter
- `sbc a, b` and `sbc b` -- WLA accepts the former
- `and a, b` and `and b` -- WLA accepts the latter
- `xor a, b` and `xor b` -- WLA accepts the latter
- `or a, b` and `or b` -- WLA accepts the latter
- `cp a, b` and `cp b` -- WLA accepts the latter

WLA accepts only one of each, giving `ERROR: Unknown symbol` for the other. What in particular keeps it from recognizing both spellings of each instruction?

vhelin, 2019-12-09:
Well, when I added support for Z80 etc. I used the best mnemonics/opcode list I could find at the time in the Internet. Perhaps the document didn't have all the aliases? Later these lists may have changed as users have reported about some missing aliases...

Anyway, if your listing here covers all the missing aliases, no problem, I can add them to opcodes_z80.c - if the alias isn't there WLA doesn't know it...

pinobatch, 2019-12-10:
BTW, these aliases should be valid for both Z80 and GB. If I get a chance over the next couple days, I'll look up what other commonly recognized aliases for Z80 and GB might be missing.

vhelin, 2019-12-10:
> BTW, these aliases should be valid for both Z80 and GB. If I get a chance over the next couple days, I'll look up what other commonly recognized aliases for Z80 and GB might be missing.

That'd be great if you have the time to do that!

vhelin, 2019-12-10:
I'll take a look at the GB-Z80 side tomorrow...

maxim-zhao, 2019-12-11:
`jp (hl)` is aliases by (the more logical) `ld pc,hl` - plus ix and iy variants.

vhelin, 2019-12-11:
Added more aliases today to GB-Z80 side (yesterday I added aliases for Z80). Something still missing?

vhelin, 2019-12-11:
> `jp (hl)` is aliased by (the more logical) `ld pc,hl` - plus ix and iy variants.

Also added aliases to these, thanks!

maxim-zhao, 2019-12-12:
Not an opcode alias but the undocumented registers `ixh` `ixl` iyh` iyl` are sometimes also referred to as `hx` `lx` `hy` `ly` respectively.

vhelin, 2019-12-13:
> Not an opcode alias but the undocumented registers `ixh` `ixl` `iyh` `iyl` are sometimes also referred to as `hx` `lx` `hy` `ly` respectively.

Thanks! I'll create `hx` `lx` `hy` `ly` versions of mnemonics that currently contain `ixh` `ixl` `iyh` `iyl`... This makes me think that the opcode parsing is pretty unoptimal in WLA DX, but at least it's very easy to add new mnemonics/opcodes to it. :)

pinobatch, 2019-12-13:
RGBASM uses `jp hl` and deprecates `jp [hl]`, in particular because it behaves as `ld pc, hl` not `ld pc, [hl]`.

vhelin, 2019-12-13:
> RGBASM uses `jp hl` and deprecates `jp [hl]`, in particular because it behaves as `ld pc, hl` not `ld pc, [hl]`.

Ok, I'll take "JP (HL)" away...

maxim-zhao, 2019-12-13:
Don't take it away! It's the official opcode, even if it is semantically wrong. Add `jp hl` (and `jp ix`, `jp iy`) as aliases.

pinobatch, 2019-12-13:
By "deprecates," I meant it accepts both but issues a warning for `jp [hl]`.

pinobatch, 2019-12-13:
Somehow the changes related to this issue have cause every single opcode to go missing from wla-z80. When I `git pull` the source code, rebuild WLA DX, install it, and run `compile.sh` in the attached zipfile, I now get errors on all three of the first three instructions in the program:
```
    di              ; disable interrupts
    im 1            ; Interrupt mode 1
    jp main         ; jump to main program
```
This produces `ERROR: Unknown symbol "di".`
Comment out the first and it instead produces `ERROR: Unknown symbol "im".`
Comment out the first and second and it produces `ERROR: Unknown symbol "jp".`

maxim-zhao, 2019-12-14:
I had the same issue in #237...

vhelin, 2019-12-14:
I needed to do "make clean" before "make" could create working executables, after I had done changes to opcodes_z80.c and opcodes_gb.c... Only after "make clean" would "make" run opcode_table_generator that would calculate new LUTs for the opcode parser...

vhelin, 2019-12-14:
Any ideas how to mod the CMake files so that it'll create Makefiles that'll run opcode_table_generator again after opcodes_*.c have been changed?

vhelin, 2019-12-14:
> Don't take it away! It's the official opcode, even if it is semantically wrong. Add `jp hl` (and `jp ix`, `jp iy`) as aliases.

Alright then, I'll add it back with a warning.

But `jp ix` and `jp iy`, are they aliases to `jp hl` in GB-Z80? The same opcode? Are `ix` and `iy` register aliases to `hl` in general? This is confusing. I didn't know GB-Z80 has `ix` and `iy` registers...

vhelin, 2019-12-14:
> Don't take it away! It's the official opcode, even if it is semantically wrong. Add `jp hl` (and `jp ix`, `jp iy`) as aliases.

Now I think I understand you: you are talking about Z80, not GB-Z80? I'll add the aliases and warnings to Z80 side of WLA...

vhelin, 2019-12-14:
Great work guys! In my eyes WLA DX has become much better than before with just a little work. :) Any other aliases that are missing?

maxim-zhao, 2019-12-14:
Please don’t make `jp (hl)` emit a warning. It's the official opcode, even if it is inconsistent in the use of `()`, I don't think it's fair to make a warning.

vhelin, 2019-12-14:
> Please don’t make `jp (hl)` emit a warning. It's the official opcode, even if it is inconsistent in the use of `()`, I don't think it's fair to make a warning.

Well, Nintendo tells us in it's official Gameboy programming guide about "JP (HL)", which is semantically incorrect. "JP HL" would be correct. And now I'm giving a warning about "JP (HL)" in WLA-GB. It would just be proper to give a warning about "JP (HL)" in WLA-Z80 as well...

vhelin, 2019-12-14:
I mean, warning is just a warning... WLA still assembles the source code if you are using "JP (HL)".

maxim-zhao, 2019-12-14:
I think the official opcodes should be supported, on Z80 that's `jp (hl)`. Supporting alternatives - as in the start of this issue - is a nice enhancement but we should not add noise to "standard" source.

vhelin, 2019-12-14:
Manufacturer documents can be sometimes buggy / wrong. Thus we now give a warning for "JP (HL)" in WLA-GB and WLA-Z80. It's still supported, though.

vhelin, 2019-12-14:
> 
> 
> Manufacturer documents can be sometimes buggy / wrong. Thus we now give a warning for "JP (HL)" in WLA-GB and WLA-Z80. It's still supported, though. Nothing is lost by doing so.

sverx, 2019-12-23:
though I'd admit that `jp (hl)` is somewhat semantically wrong, I wouldn't issue any warning for an opcode that's historically always been written that way (also, it would be totally wrong if it was `ld pc,(hl)`, but the way it was written it meant 'jump to code in memory at address pointed by HL' which makes it less a mistake...)

Kroc, 2019-12-23:
It's not right, but it is what's output by pretty much all existing disassembly tools!

pinobatch, 2019-12-23:
I've been convinced. Now I'd recommend that the warning for mnemonics that are official but misleading (such as `jp (hl)` in Z80 and GB) should be off by default. Do the big toolchains have anything like `-Wmisleading-mnemonics`?

vhelin, 2019-12-23:
Just tested RGBDS assembler for GB-Z80 and it just gives syntax error if I try to use the official "JP (HL)". Instead it takes "JP HL". :)

Kroc, 2019-12-23:
Perhaps consider a "-wpedantic" like option that actually turns on style and suggestion warnings.

pinobatch, 2019-12-24:
Though that might be better if split out as a separate issue.

vhelin, 2021-04-26:
@pinobatch I totally missed your commit about "CPL A" and "STOP $01" initially, but saw it today and made the changes to the main branch of WLA DX. I guess we can close this issue now?

Kroc, 2021-04-27:
I think you can close this one, and if there are any more missing instruction forms a new issue can be opened.

closed 2021-04-27 by vhelin