vhelin/wla-dx #272: Emit diagnostic for all errors, not just one

pinobatch, 2019-12-09:
I put this code in a Z80 program:
```
  inc fg
  inc jk
```
You and I both know that Z80 has no register pairs by those names. But instead of giving errors about both lines, wla-z80 gives `ERROR: Unknown symbol "inc".` on only one of the two lines. This breaks the work flow of assembling, fixing more than half the errors, and assembling again.

vhelin, 2019-12-09:
I'm not totally sure what kind of changes this would requite to WLA so I'm thinking about postponing this a little... Perhaps read the source code when I have extra free time and try to investigate.

vhelin, 2021-03-19:
For the record, we exit right away everywhere when we spot an error, so showing all the errors for every problematic line in source code during one run of WLA would require a lot of changes all over the place.

vhelin, 2023-02-21:
I'm not sure if this is worth the trouble. I mean after the 1st error the state of the parser might be wrong and it might then emit error messages about errors that don't exist, because the state is wrong. I'm not touching this not at least now.

vhelin, 2023-02-21:
Well, if you can make the parser(s) emit only real errors and all of them at once, feel free to implement. :)

vhelin, 2023-07-16:
As a side note: I've worked on a WPF/C# project this year at work and there if one code line has an error I get 30+ errors in the console and need to search for the real error that's there, it's rarely the topmost, which takes sometimes a lot of time and often it's just better to revisit the recent the I've written and double check it. So I'm not a big fan of adding support for this feature as it might lead to other problems...

Unless in the case of an error we skip the whole line where the error is, assume that either a whole line of code works or it's just that line which is broken. Hmm, I might give this a try.

vhelin, 2023-07-17:
While testing -c I encountered something that's going to water this feature down a little.

WLA-GB knows instruction "AND (HL)", but also "AND [number]". So if I write "AND (BC)" then WLA thinks "(BC)" is just a label inside parentheses i.e., plain BC (label) thus matching "AND [number]" and WLA doesn't think it's a mistyped instruction. WLALINK later tries to find label BC and gives an error on that line when it doesn't find BC... So -c doesn't help you with such instructions. Not sure what to do about it or if it can be easily made better.