vhelin/wla-dx #274: Assembler fails if struct path is longer than 63 characters

SavagePencil, 2019-12-29:
Assume we have the following structures:

```
.STRUCT demo
    ABCDEFGHIJKLMNO  DB
    ABCDEFGHIJKLMNOP DB
.ENDST

.STRUCT ABCDEFGHIJKLMN
    abcdefghijklmnopqrstuvwxyzabcdef INSTANCEOF demo
.ENDST

```

The following will assemble correctly:
`            ld      (iy + ABCDEFGHIJKLMN.abcdefghijklmnopqrstuvwxyzabcdef.ABCDEFGHIJKLMNO), $00`

...while this one results in `ERROR: Unknown symbol "ld"`:

`            ld      (iy + ABCDEFGHIJKLMN.abcdefghijklmnopqrstuvwxyzabcdef.ABCDEFGHIJKLMNOP), $00`

It would appear that the second instance goes to 64 characters being added to IY, while the first is at exactly 63.  Can the buffer size be extended?

vhelin, 2020-01-03:
A fix to this will come today evening. The stack calculator thought that all strings are max 63 bytes long (a very old restriction). Made it to use MAX_STRING_LENGTH instead, which is 255 bytes currently...

vhelin, 2020-01-03:
Can you try out the latest sources? Will they fix this problem?

SavagePencil, 2020-01-03:
Confirmed this as working with latest!

closed 2020-01-03 by SavagePencil