vhelin/wla-dx #275: Can't .DB a negative value inside a .DSTRUCT

SavagePencil, 2020-01-01:
I'm unable to get a negative value to instantiate for a .DB inside of a .DSTRUCT

If I do the following:

```
.STRUCT MyStruct
    ByteVal DB
.ENDST

.DSTRUCT defStruct INSTANCEOF MyStruct VALUES
    ByteVal .DB -1
.ENDST
```

I get the following error:
```
Unknown directive ".DB-1".
ERROR: Couldn't parse ".DB-1".
```

As a workaround, I found that I can do the following:
```
.DSTRUCT defStruct INSTANCEOF MyStruct VALUES
    ByteVal .DB (-1)
****OR****
    ByteVal .DB $FF
.ENDST
```

...but that's kinda lame.

vhelin, 2020-01-03:
A fix to this will come later this evening. The preprocessor takes the space character away between ".DB" and "-1" and the parser cannot handle such situations. Will make the parser better...

vhelin, 2020-01-03:
Can you try out the latest sources? Will they fix this problem?

SavagePencil, 2020-01-03:
Confirmed this as working now!

closed 2020-01-03 by SavagePencil