vhelin/wla-dx #276: Can't create .DSTRUCT as a child of a label with the same name as another child of a DIFFERENT label.

SavagePencil, 2020-01-03:
This is a difficult bug to write out, but it is easier to see in person:

```
.STRUCT MyStruct
    ByteVal         DB
.ENDST

Label_A:
@Child:     ; This label works (it is Label_A@Child)
.DSTRUCT @Instance INSTANCEOF MyStruct VALUES
    ByteVal         .DB $FF
.ENDST
```
This works fine; I have `Label_A@Child` and `Label_A@Instance`.

However, when I add the following:
```
Label_B:
@Child:     ; This label works (it is Label_B@Child)
.DSTRUCT @Instance INSTANCEOF MyStruct VALUES
    ByteVal         .DB $A5
.ENDST
```

This fails with an error `DIRECTIVE_ERROR: "_sizeof_@Instance" was defined for the second time.` and `ERROR: Couldn't parse ".DSTRUCT".`

I would expect that this creates four labels:
```
1. Label_A@Child
2. Label_A@Instance
3. Label_B@Child
4. Label_B@Instance
```

However, it only creates the first three, and sees a conflict between 2 and 4.

Calindro, 2020-03-25:
Unfortunately, the whole `_sizeof_` stuff seems to be a huge mess. Since its introduction we frequently received errors like yours. And we weren't even using any special features of the assembler. All we used were plain sections and plain labels. Due to its unpredictable nature we also couldn't find any reliable workaround for it.
As a result we went back to the last version right before the introduction of `_sizeof_`.
Just recently, we wanted to give it a try again and actually tried to make use of them but they didn't even seem to get defined reliably and without using them they still produced errors like this one. So again we reverted back to that old version. I think it's a shame because that version is more than a year old already and thus misses all of the recent bug fixes and improvements.
Considering that these labels enjoy to cause harm and don't even seem to get defined reliably, I don't see much use for them.
@vhelin Ideally you could make them so that they only get defined if they are used? Or make them optional so that if anyone happens to want to give them a chance they can enable them? It can be really frustrating when something that isn't under your control keeps you from producing a working rom.

vhelin, 2020-03-25:
Ok, I'll work on this next as it seems to cause bigger problems to users... Hopefully I can find a fix to this this week.

vhelin, 2020-03-31:
Ok, I hope my commit fixes both of the issues reported here in this thread.

SavagePencil, 2020-04-11:
I don't think this is fully fixed.  I have the following:

```
@TileSets:
.DW @@DefaultFont
.DW @@LevelSet_4bpp

;======DEFAULT FONT TILE SET======
@@DefaultFont:
.DSTRUCT @@@TileSetDescriptor INSTANCEOF sTileSetDescriptor VALUES
    Flags:      .DB TILESET_FLAGS_1BPP_UNCOMPRESSED | TILESET_FLAGS_STRUCTURE_BLOCK | TILESET_FLAGS_BLOCK_HAS_DEST_INDEX | TILESET_FLAGS_HAS_1BPP_REMAP
    NumTiles:   .DW ( DefaultFont_1bpp_DataEnd - DefaultFont_1bpp_DataBegin ) >> 3

    ; 1bpp-Only Pal Remaps:
    PaletteRemap_1bpp.Pal0_Remap:   .DB 0
    PaletteRemap_1bpp.Pal1_Remap:   .DB 1
.ENDST
.DSTRUCT @@@TileData INSTANCEOF sTileSet_BlockEntry VALUES
    DestTileIndex:  .DW GFX_TILE_INDEX_FONT
    pData:          .DW DefaultFont_1bpp_DataBegin
.ENDST

;======LEVEL BACKGROUND TILE SET======
@@LevelSet_4bpp:
.DSTRUCT @@@TileSetDescriptor INSTANCEOF sTileSetDescriptor VALUES
    Flags:      .DB TILESET_FLAGS_4BPP_UNCOMPRESSED | TILESET_FLAGS_STRUCTURE_BLOCK
    NumTiles:   .DW (BMP2Tile_4bpp_Data@End - BMP2Tile_4bpp_Data@Begin ) / _sizeof_sTile
.ENDST
.DSTRUCT @@@TileData INSTANCEOF sTileSet_BlockEntry VALUES
    DestTileIndex:  .DW $80
    pData           .DW BMP2Tile_4bpp_Data
.ENDST
@@@@End:
```

I'm receiving the error:

```
DIRECTIVE_ERROR: "_sizeof_@@@TileSetDescriptor.PaletteRemap_1bpp" was defined for the second time.
```

It looks like it's getting confused between 

`@TileSets@DefaultFont@TileSetDescriptor.PaletteRemap_1bpp` 

...and:

`@TileSets@LevelSet_4bpp@TileSetDescriptor.PaletteRemap_1bpp` 

Here is what the sTileSetDescriptor struct looks like:

```
.STRUCT sTileSetDescriptor
    Flags               DB      ; Composite of TILESET_FLAGS_*
    NumTiles            DW      ; How many tiles in this set?

    ; Type-based flag values
    .UNION PaletteRemap_1bpp
        Pal0_Remap      DB      ; Color remap for 0s in 1bpp
        Pal1_Remap      DB      ; Color remap for 1s in 1bpp
    .ENDU
.ENDST
```

SavagePencil, 2020-04-11:
(yes I realize it's odd that I have a union section with only one entry; it's placeholder for other tile sets to expand on later)

vhelin, 2020-04-11:
Uuh, forgot to handle unions... How does it work now?

SavagePencil, 2020-04-11:
It works!

closed 2020-05-13 by vhelin