vhelin/wla-dx #277: Assembler can't get bank of 2 numbers added together

oziphantom, 2020-01-18:
So I was using 9.8 which just put a really wrong value into the ROM, so I've upgraded to 9.10 which now just throws a really funny error. 
wla-65816
Given
```
;LoadBlockToVRAM SRC_ADDRESS, DEST, SIZE
;   requires:  mem/A = 8 bit, X/Y = 16 bit
.MACRO LoadBlockToVRAM
    lda #$80
    sta $2115
    ldx #\2         ; DEST
    stx $2116       ; $2116: Word address for accessing VRAM.
    lda #:\1        ; SRCBANK
    ldx #\1         ; SRCOFFSET
    ldy #\3         ; SIZE
    jsr LoadVRAM
.ENDM
```

if one calls this with 
```
.EQU	Intro_greyText2nd = Intro_greyText+$1c00
	LoadBlockToVRAM Intro_greyText2nd,$4e00,$1c00
```
I.e I have a binary file Include somewhere down in the banks, and I need to break the transfer into 2, so I offset halfway and then copy half of it. You get `COMPUTE_PENDING_CALCULATIONS: Result (56320/$dc00) of a computation is out of 8-bit range.` where line mentioned is `lda #:\1        ; SRCBANK` 
which raises 2 questions. 
1.) Why does it think it can't hold 1 in 8bits
2.) Why is it doing a DC00/DC00 to work out the bank? 

I have also tried `.EQU	Intro_greyText2nd = (Intro_greyText+$1c00)` same result and `.EQU	Intro_greyText2nd = (Intro_greyText+1c00h)` just in case it was an odd $ form rather than this assemblers preferred h form.

vhelin, 2020-01-19:
It seems that getting the bank number using : from a definition (or plain address value) doesn't work. If you use : with a plain label then it works... I'll look what's gone wrong in WLA...

vhelin, 2020-01-22:
A question: If the address "intro_greyText + $1c00" crosses a slot border, so it's not in the same slot where "Intro_greyText" is, how should WLA know what bank is loaded into that slot at the time of running that code?

oziphantom, 2020-01-23:
It should resolve the address to full 24bits then take the bank from said address. 
So if you have somewhere that points to $00ff00 and you add $100 to it you would get $010000 and thus the bank is $01 
If an included file escapes a slot/bank I would expect the emulator to at least warn me, if not error - does it not?

The case I have is access data in the ROM and the range is withing the size of the included binary data. This is not accessing RAM.

vhelin, 2020-01-26:
I'm currently working on this, but getting this to work actually requires a lot of changes in various places... :P

aexoden, 2020-01-31:
I'm not sure if it's related to these changes (and if not, I can open a new issue), but the newest version seems to be having trouble giving the bank of 24-bit addresses I have defined in a RAMSECTION. (It's returning $00 instead of the desired $7E or $7F.) I also have another issue that's stranger and I'll open an issue for that.

aexoden, 2020-02-02:
The latest code is still leaving me unable to get the bank number of variables declared in a RAMSECTION like the following:
```
.ramsection "LowRAM" bank $7E slot 0
    test_var db
.ends
```

If this isn't related and you'd prefer I open a second issue, let me know.

vhelin, 2020-02-02:
Sorry, I missed this one... Hmm, no need to open a new issue, I can start looking for the bug with this information...

vhelin, 2020-02-02:
Ok, I'm gettig pretty tired here. I hope I was able to fix the issue. If not, please let me know and I'll take a look at this tomorrow...

oziphantom, 2020-02-03:
I was able to work out finally how to cross build it on windows. And I have confirmed that labels with + XXXX now get the right bank in the case I posted above. Thanks. I will leave this open due to the new issue posted by aexoden.

aexoden, 2020-02-03:
The new issue I brought up is also now working correctly.

closed 2020-02-04 by oziphantom