vhelin/wla-dx #279: WLALINK: Support C64 PRG output

Kroc, 2020-01-21:
I'm working on a Forth-like scripting language, primarily for the C64 with the intention of porting it to the 65C02-based Commander X16. I'd like to also make Z80 ports though, and only WLA-DX can offer me the one assembler that can use the same code-base to support 65XX & Z80 systems.

I know that WLA-DX is geared up for ROM development on consoles, but I'd like to help with improving the micro-computer support. I'll probably filing a number of bugs as I go.

Firstly though, I need to build C64 .prg files. A .prg file (for a C64) has a two-byte header, the load-address, in little-Endian order.

Could you please add an option to WLALINK to output a binary with C64 prg header using the first address used, or a provided load-address.

Since I intend to port to other systems too, and more specific binary format may be needed, it perhaps would be best to implement a proper "type" argument, e.g.

```
wlalink -t C64 # outputs a C64 .prg file with load-address automatically determined
wlalink -t C64 -a 0x0801 # as above, but with manual load-address
```

vhelin, 2020-01-21:
Sounds cool! Let's see what I can do at the end of this week (if not sooner)...

Kroc, 2020-01-21:
FYI here's the workaround that I'm currently using to get a valid header:

```
.DEF basic_start = $0801

.SLOT   2
.ORGA   basic_start-2
.WORD   basic_start
.ORGA   basic_start
```

vhelin, 2020-01-22:
Does it work? If you don't give WLALINK -a then it'll use the address of the first non-empty byte (first byte of the outputted program) in the output file as the load address. Is this ok?

Kroc, 2020-01-22:
I'm working out how to compile WLA-DX for Windows ATM, I'll probably get back to you tomorrow about results!

Kroc, 2020-01-22:
Okay, worked! That makes C64 development easier!

Kroc, 2020-01-22:
Please consider setting up some kind of CI GitHub integration so that every commit gets built; compiling WLLA-DX for Windows is very, very difficult

Kroc, 2020-01-23:
This might be a bug. If I compile the following code

```
; C64 memory map:
;
.MEMORYMAP
        DEFAULTSLOT     2
        SLOTSIZE        $00FD
        SLOT 0          $0003   ; zero-page (variable space)
        SLOT 1          $0100   ; 6502-stack (variable space)
        SLOTSIZE        $FE00
        SLOT 2          $0200   ; main program RAM ("ROM")
.ENDME

.ROMBANKSIZE            $FE00
.ROMBANKS               1

.SLOT   2
.ORGA   $0801

-       inc $d020
        jmp -
```

WLA-LINK will use a starting address of $0601 automatically! Using `-s 0801h` fixes this, but the "first-address" calculation is wrong somehow!

vhelin, 2020-01-23:
> 
> 
> Please consider setting up some kind of CI GitHub integration so that every commit gets built; compiling WLLA-DX for Windows is very, very difficult

I have no idea about how to do that, but I can take a look...

vhelin, 2020-01-23:
> 
> 
> This might be a bug. If I compile the following code
> 
> ```
> ; C64 memory map:
> ;
> .MEMORYMAP
>         DEFAULTSLOT     2
>         SLOTSIZE        $00FD
>         SLOT 0          $0003   ; zero-page (variable space)
>         SLOT 1          $0100   ; 6502-stack (variable space)
>         SLOTSIZE        $FE00
>         SLOT 2          $0200   ; main program RAM ("ROM")
> .ENDME
> 
> .ROMBANKSIZE            $FE00
> .ROMBANKS               1
> 
> .SLOT   2
> .ORGA   $0801
> 
> -       inc $d020
>         jmp -
> ```
> 
> WLA-LINK will use a starting address of $0601 automatically! Using `-a 0801h` fixes this, but the "first-address" calculation is wrong somehow!

That's because the RAM address 0801h is in your case the ROM address 0601h as you insert that ROM bank into slot 2 that starts at address 0200h... At the moment WLALINK uses the ROM address of the first used byte as the automatic load address.

This is actually a tricky situation: WLA doesn't save the used RAM addresses for random bytes. Only labels have that information... Would it be ok if -t would require -a as well as -a can handle labels?

Kroc, 2020-01-23:
That would be fine, with the one exception that it means that I have to duplicate the load-address in two different disconnected environments -- in the Batch script, and in the Assembler source. This makes things more prone to fault if the address is changed in one place and not the other; something easily overlooked by a new user to the codebase. There will be ways to work around this, I'm sure, but I'm always looking for a more elegant solution.

One possibility is having a directive for setting the PRG address; I.e.

```
.PRG some_const ; use an existing constant or label name
.PRG * ; set the .prg address to the current absolute PC
```

Kroc, 2020-01-23:
I see the new commits, will compile these and test.

vhelin, 2020-01-23:
.PRG (or perhaps .C64PRG would be more appropriate?) sounds like a nice additional feature, I can do that...

Kroc, 2020-01-23:
I don't know if other systems use the same file header, I need to look into that for ports to other systems.

Kroc, 2020-01-23:
Those commits fixed the problem! With a SLOT address of $0200, the PRG address is automatically detected as $0801 👍

vhelin, 2020-01-23:
You need to have some label there. In your case you have '-' there and WLALINK finds it's RAM address... :)

Kroc, 2020-01-23:
Suggestion: rename "c64PRG" &c. to "CBM", as this header format is used by all Commodore 6502-based machines; the PET, VIC20, C64, C16 & Plus/4 and more.

vhelin, 2020-01-23:
CBMPRG might be better as on some other platforms there might be PRG file formats?

Kroc, 2020-01-23:
The X16 uses PRG files, but in the same format (it's BASIC-compatible with the C64). I can't speak of other microcomputers yet, but I intend to look further down the line. The file name is what needs the ".prg" so the user has already set that with `-o`, and a output type of `CBM` would imply a ".prg" file extension anyway. I don't think you need to be overly cautious.

Kroc, 2020-01-23:
A number of different systems have tools for putting binaries onto disk; meaning that it doesn't need to be done in the assembler. For Amstrad CPC, this appears to be the AMSDOS header (http://www.cpcwiki.eu/index.php/AMSDOS_Header), but they have multiple existing tools to put binaries into disk images and generate the header for you.

ZXSpectrum uses TAP (tape) files, and also Amstrad disk (same as above). ".SNA" ("snapshot") files are RAM dumps, used often in digital distribution due to skipping tape loading: http://rk.nvg.ntnu.no/sinclair/faq/fileform.html#SNA

I wouldn't worry about supporting these formats yet as you'll need someone making use of them to test them. I'll file bugs when I get to the relevant ports :)

vhelin, 2020-01-23:
X16 == Commander X16? Never heard of that before today. Does it mean that X16 uses the CBM PRG file format, or X16's own PRG file format is the same as CBM PRG file format?

vhelin, 2020-01-23:
Is the file format called X16 PRG or CBM PRG in the X16 documentation / community?

Kroc, 2020-01-23:
Just "PRG", nothing else. The machine is designed to be Commodore-like, so it uses most of its formats and even a partially compatible KERNAL.

vhelin, 2020-01-23:
It seems that Spectrum also has a PRG file format:

  http://rk.nvg.ntnu.no/sinclair/faq/fileform.html

It's not obvious that .PRG is a CMB file. And it doesn't hurt to have that extra "CBM" typed in with "PRG" currently in WLALINK... :) At least everybody knows then what it is.

If X16 supports CBM PRG files, when just use CBM PRG... ?

Kroc, 2020-01-23:
You're putting too much emphasis on filename extension over header format?
The spectrum PRG format is a snapshot file, not the same as a .prg file containing a CBM header.
The filename is not important, as the user decides that. The header format though is the critical thing. A CBM header is such that is used in files intended for CBM machines (PET, VIC, C64 et al)

But sure, use "CBMPRG" (or "CBM_PRG"), it's not going to matter, I just want to ensure we're talking about the same thing here :P

vhelin, 2020-01-23:
I'm just thinking about how we should name the header support so that it would be 100% obvious right away what it is to the person who reads the docs / command line helps...

Just PRG is a little bit vague according to the latest information, but CBM PRG should be instantly obvious?

Kroc, 2020-01-23:
Go with CBM PRG; if the X16 devs decide to deviate in the future it could be named X16_PRG.

closed 2020-03-10 by vhelin