vhelin/wla-dx #281: Placing a `.RAMSECTION` after any code, breaks 8-bit relative references

Kroc, 2020-01-23:
```
.SLOT 2
-       lda $00
        beq -

.RAMSECTION "bug" SLOT 0
        bug BYTE
.ENDS

loop:   lda $00
        beq loop
```

Assembly will fail, believing the branch to be too far!

vhelin, 2020-01-23:
If you now issue

  .RAMSECTION "bug" SLOT 0 RETURNORG

it'll work...

Kroc, 2020-01-24:
But that shouldn't be necessary to prevent outright breakage of relative addressing? There can't be any old code in use that relies on .RAMSECTIONs being defined after some code, because every relative address would fail to assemble!

vhelin, 2020-01-24:
That's actually true. But if I change .RAMSECTION to do RETURNORG by default, then I would also have to chage the normal .SECTION to do the same as it'd be weird if these two worked differently. And there might be code out there that depends on the .SECTIONs not to do RETURNORG by default... Perhaps?

vhelin, 2020-01-24:
I could also try to investigate why .RAMSECTION messes up relative addressing...

closed 2020-03-10 by vhelin