vhelin/wla-dx #282: Bad auto-detection of operand size

Kroc, 2020-01-23:
The 6502 instruction `lda` can take both 8, and 16-bit operands, however it appears to be over-zealous in the case of an obviously 16-bit operand being expected and perfectly okay.

The following code:

```
.ORGA $0801

print:
        ldx # 0
-       lda @ver, x
        jsr KERNAL_CHROUT
        inx
        cpx # 7
        bne -

@ver:   .ASC    "Pling!", 13
```

Produces the error message:
```
FIX_REFERENCES: Value ($82e) of "init@ver" is too much to be a 8-bit value.
```

The `@ver` reference is 16-bit and shouldn't need a hint as this is perfectly acceptable for the `lda` instruction. In fact, if `@ver` was in the zero-page, an 8-bit operand should be used for speed and size optimisation. No error message is needed in either situation.

The only instance, where you would need to use `lda` with a 16-bit operand, using a value <$100 would be where the exact number of cycles or bytes are required, and the edge case can easily be handled by using `.w`.

vhelin, 2020-02-09:
As a quick fix, please use .16BIT at the beginning of your source file. The WLA assembler currently defaults to 8-bit arguments...

Kroc, 2020-02-10:
Not an option for 6502s; as all zero-page opcodes are changed to slower, larger 16-bit opcodes! My program goes from 1'545 bytes to 1'670 bytes and costs 2-3 more cycles per instruction referencing the zero-page! Don't worry, adding `.l` here and there occasionally is no problem :)

cr1901, 2023-07-20:
I'm also running into this issue for I/O (contrived example for [WDC's board](https://wdc65xx.com/Single-Board-Computers/w65c02sxb/), `load.asm` is negligible):

```
.MEMORYMAP
DEFAULTSLOT 1
SLOT 0 START $0000 SIZE $8000
SLOT 1 START $8000 SIZE $8000
.ENDME

.ROMBANKSIZE $8000
.ROMBANKS 3

.RAMSECTION "ZERO_START" BANK 0 SLOT 0 ORG 0
foo db
.ENDS

.RAMSECTION "STACK_PAGE" BANK 0 SLOT 0 ORG $100
. DSB 256
.ENDS

.RAMSECTION "MAIN_RAM" BANK 0 SLOT 0 SEMIFREE ORG $200
bar db
.ENDS

.RAMSECTION "PERIPHERALS" BANK 0 SLOT 0 FORCE ORG $7f00
BUS0 DSB 32
BUS1 DSB 32
BUS2 DSB 32
. DSB 32
ACIA DSB 32
PIA DSB 32
VIA DSB 32
TIDE DSB 32
.ENDS

.BANK 0 SLOT 1
.ORG $0000

.BANK 0 SLOT 1
.SECTION "Entry" ORG 0
lda ACIA
.ENDS
```

```
PS C:\msys64\home\William\Projects\legacy\6502\W65C02SXB\hello> make -B
wla-65c02 -i -k -o hello.o hello.asm
wla-65c02 -i -k -o load.o load.asm
wlalink -i -S linkfile hello.bin
hello.o: hello.asm:39: FIX_REFERENCES: Value ($7f80) of "ACIA" is too much to be a 8-bit value.
make: *** [Makefile:4: hello.bin] Error 1
```

I think `lda ACIA` should "do the right thing" and create a 16-bit absolute load.

vhelin, 2023-07-21:
Yeah, in this case it should be possible to see that ACIA is inside a .RAMSECTION that's forced to a 16-bit address and that ACIA is a 16-bit value. Adding this logic would require new code to be added to every such instruction parser that's first processing an 8-bit version of an instruction that's followed by a a processor of the same but 16-bit version of that instruction. Doable, yes. Trivial, no.