vhelin/wla-dx #287: Allow SLOT IDs > 255, (or label name/string)

Kroc, 2020-01-30:
SLOT IDs are rather arbitrary, especially on a system where there will usually be overlapping SLOTs; that is, you may wish to assign RAM sections into SLOTs that overlap an underlying BANK/SLOT. Take for example this commodore 64 map:

```
.MEMORYMAP
        DEFAULTSLOT     2
        SLOTSIZE        $00FD
        ; $00/$01 are not available on
        ; the C64 as these are the CPU port
        SLOT 0          $0002   ; zero-page (variable space)
        SLOT 1          $0100   ; 6502-stack (variable space)
        SLOTSIZE        $FE00
        SLOT 2          $0200   ; main program RAM ("ROM")

        SLOTSIZE        $1000
        SLOT 3          $C000
        SLOT 4          $D000
        SLOTSIZE        $2000
        SLOT 5          $E000
.ENDME
```

Since the C64 is all RAM, you can begin your code at $0200 (although in practice, you would begin at $0801) and continue going all the way up to $FFFF

However, if you want to assign some large areas of RAM for non-code (such as where game levels are loaded in), you don't want to assign that space _inline_ with your code. `$C000-$CFFF` is an unused patch of RAM and is often used for variables or machine code. $D000-$DFFF is the I/O address space, and no code can be placed there, but the RAM underneath can be used. $E000-$FFFF is the KERNAL ROM space, but again the RAM underneath can be used.

In this instance, the SLOT numbers are not helpful or indicative when used on `.SECTION`/`.RAMSECTION` directives. Being able to assign more indicative numbers to SLOT IDs would go some way in increasing readability:

```
.MEMORYMAP
        DEFAULTSLOT     2
        SLOTSIZE        $00FD
        ; $00/$01 are not available on
        ; the C64 as these are the CPU port
        SLOT $00        $0002   ; zero-page (variable space)
        SLOT $100       $0100   ; 6502-stack (variable space)
        SLOTSIZE        $FE00
        SLOT $0200      $0200   ; main program RAM ("ROM")

        SLOTSIZE        $1000
        SLOT $C000      $C000
        SLOT $D000      $D000
        SLOTSIZE        $2000
        SLOT $E000      $E000
.ENDME

; i.e.
.RAMSECTION "level_data" SLOT $C000
        ; ...
.ENDS
```

For projects compiling for multiple systems, that may be assigning code & RAM sections into vastly different memory layouts; slot numbers work against us. What might be SLOT 3 on one system, might be SLOT 5 on another. Having some kind of either label (or string) name assignment for SLOTs would alleviate this concern:

```
.MEMORYMAP
        DEFAULTSLOT     2
        SLOTSIZE        $00FD
        ; $00/$01 are not available on
        ; the C64 as these are the CPU port
        SLOT lovars     $0002   ; zero-page (variable space)
        SLOT stack      $0100   ; 6502-stack (variable space)
        SLOTSIZE        $FE00
        SLOT main       $0200   ; main program RAM ("ROM")

        SLOTSIZE        $1000
        SLOT hivars     $C000
        SLOT io         $D000
        SLOTSIZE        $2000
        SLOT hiram      $E000
.ENDME
```

vhelin, 2020-01-30:
A string label for a slot is a great idea! But I think I can also make it so that when you refer to a SLOT you can use its address. But when you have a SLOT starting at address $2 and you try to use SLOT 2, that should issue a warning and a note which one is used, a SLOT starting at $2 or a SLOT number 2...

Kroc, 2020-01-31:
Agreed!

vhelin, 2020-02-01:
I made it so that SLOTs can now be given names, but you still need to give a SLOT a number as well. Next I'll add support for SLOT selection by its address...

vhelin, 2020-02-01:
I hope it works now, it was quite a big change in the code...

Kroc, 2020-02-01:
This will be great for sharing sections between different systems!

closed 2020-03-10 by vhelin