vhelin/wla-dx #288: [wla-65816] Incorrect calculation results under certain conditions

aexoden, 2020-01-31:
Given the following test program (which is slightly convoluted, but does represent something the code I'm working with is doing):

```
.memorymap
	slotsize $8000
	defaultslot 1
	slot 0 $0000
	slot 1 $8000
.endme

.lorom

.rombanksize $8000
.rombanks 3

.bank 0 slot 1
.section "Bank0"
main:
	.dw :test_data << 12
	.dw (test_data - $8000) >> 3
	.dw (:test_data << 12) + ((test_data - $8000) >> 3)
.ends

.bank 2 slot 1
.section "Bank2"
	.dsb $818 $00
test_data:
	.db $FF, $FF, $FF, $FF
.ends
```
This example does compile, but the binary output I'm getting (in the first three bytes) is 
```
00 20   03 01   03 31
```
I would expect the third word to be the sum of the first two, but there is clearly something else going on. ($2000 + $0103 != $3103) (In my actual code, the summed values end up so high they fall out of the 16-bit range and therefore it can't compile at all.)

This might not be the simplest test case possible, but I'm not familiar enough with wla to try to isolate it.

aexoden, 2020-02-01:
After bisecting, 0774d985c259a9567446d5fb38ca24863b9e48d1 is the first commit with the problem.

vhelin, 2020-02-01:
Thanks for the heads up! I'll try to fix this asap...

vhelin, 2020-02-02:
Found the culprit: my new code changed all addresses to ROM addresses in a computation if a ':' was used. In your third computation we are mixing both a ROM address (for ':') and RAM address... Fixing this will take some time. :P

vhelin, 2020-02-02:
Ok, I hope the latest sources work again for you. I added test cases for the calculations you are using so we can see automatically in the future if this one breaks again... Thanks!

aexoden, 2020-02-02:
The newest code does seem to fix this particular issue. Thanks.

closed 2020-02-02 by aexoden