vhelin/wla-dx #289: CADDR not working?

Kroc, 2020-02-05:
Using this setup,

```
.DEF    prev_term       $0000

.MACRO  .term   ARGS    str
        .WORD   prev_term
        .REDEF  prev_term       CADDR - 2
        .BYTE   str.length, str, 0
.ENDM
```

the word that gets written to the file after each call of the macro is always `$FFFE`;
it appears that `CADDR` is returning '$0000' in this context?

Kroc, 2020-02-05:
(This is a critical bug for me because I can't create a backward linked-list, and therefore I cannot produce a working program at all!)

vhelin, 2020-02-05:
I'll try to take a look at this today evening. I guess my latest changes to the stack calculator in WLALINK broke this and that? I'll add this test also to our test database so it doesn't break again in the future. :)

vhelin, 2020-02-05:
Actually I think this hasn't worked for a long time if ever. Here we have CADDR inside stack calculations inside definitions, and definition's don't have addresses thus the calculations failed... I made it so that when we calculate the pending computation's we'll set a global var to hold the current output address for CADDRs inside definition stack calculations.

Kroc, 2020-02-05:
Awesome, thanks for tackling this so fast, I'll be testing this right-away!

Kroc, 2020-02-05:
This is still producing incorrect results! I changed the macro to create a label with each invocation:
```
.MACRO  .term   ARGS    str
__\._\@:
        .WORD   prev_term
        .REDEF  prev_term       CADDR - 2
        .BYTE   str.length, str, 0
.ENDM
```
Here's a comparison of what the label is in the symbol file, next to the value found in RAM on the C64 at that point:

```
00:0a30 __.term_0 ;=$0000, as expected
00:0a3d __.term_1 ;=$08B3, should be $0A30!
00:0a48 __.term_2 ;=$0864, should be $0A3D!
00:0a5f __.term_3 ;=$085D, should be $0A48!
...
```

Kroc, 2020-02-05:
SLOT address seems to be a partial factor, most of these addresses are off by almost $200 -- the starting SLOT address used by my C64 memory map

vhelin, 2020-02-05:
My testcase was too simple and I didn't see that I used the ROM address for CADDR instead of RAM address... Does it work now better?

vhelin, 2020-02-05:
Oops, what happened to list files? :D Uuuh...

Kroc, 2020-02-05:
Ah, I see a different kind of problem now; the value of `prev_term` appears it is being set to CADDR-2 *before* it gets written to the output -- i.e. the backlinks are linking to themselves -2

```
MACRO  .term   ARGS    str
\.\@:   .WORD   prev_term    ; < is writing CADDR-2 (from below), not the *previous* value
        .REDEF  prev_term       CADDR - 2
        .BYTE   str.length, str, 0
.ENDM
```

RE: listfiles, I get no list file output at all. They're never produced despite the command switch being there

vhelin, 2020-02-05:
To get a list file out of WLALINK you need to give "-i" to the assembler as well...

vhelin, 2020-02-05:
> Ah, I see a different kind of problem now; the value of `prev_term` appears it is being set to CADDR-2 _before_ it gets written to the output -- i.e. the backlinks are linking to themselves -2
>
> ``` [...]

Are you sure about this?

In my output of this:

...

.memorymap
        slotsize $8000
        defaultslot 1
        slot 0 $0000
        slot 1 $8000
.endme

.lorom

.rombanksize $8000
.rombanks 3

.DEF    prev_term       $0000

.MACRO  .term   ARGS    str
__\._\@:
        .WORD   prev_term
        .REDEF  prev_term       CADDR - 2
        .BYTE   str.length, str, 0
.ENDM

.bank 0 slot 1
.section "Bank0" force returnorg
main:
        .db "01>"
        .term "YES"
        .db "<01"

        .db "02>"
        .term "NO"
        .db "<02"

        .db "03>"
        .term "HELLO"
        .db "<03"
.ends

...

... the first ".WORD prev_term" is set to $00 $00 as expected...

Kroc, 2020-02-05:
> to get a list file out of WLALINK you need to give "-i" to the assembler as well...

https://github.com/Kroc/pling/blob/master/build.bat#L5-L21

Kroc, 2020-02-05:
The first, yes, but not the ones afterwards:
```
00:0a30 .term0 ;=$0000, as expected
00:0a3d .term1 ;=$0A3B = self -2, should be $0A30
00:0a48 .term2 ;=$0A46 = self -2, should be $0A3D
...
```

vhelin, 2020-02-05:
Ok, you want the CADDR of the place where you issue .REDEF, not the CADDR of the place where CADDR is used. This might be a problem as we would need to know CADDR in the assembler so it could assign it to the definition .REDEF creates... But WLALINK is the one who knows all the addresses.

This is not something I can fix today. Can you make your .MACRO someway else? Perhaps using that __\._\@ label instead of CADDR? Something like that?

Kroc, 2020-02-05:
I don't think we can build labels dynamically; I can define a new label for each macro call, but from within that macro, we can't create reference a label whose name contains a calculation like `\@-1`

vhelin, 2020-02-05:
Perhaps I could emit an internal label every time CADDR is used and use its RAM address? I'll need to think about this...

vhelin, 2020-02-05:
Hmm perhaps I could help by adding support for (something like) \£ which would be \@-1? That's a workaround, but sounds useful...

Kroc, 2020-02-05:
> Ok, you want the CADDR of the place where you issue .REDEF, not the CADDR of the place where CADDR is used.

That's not quite right; the value of `prev_term` should be written to the object file, before `prev_term` gets redefined (it doesn't matter than it contains `CADDR`). The problem we're seeing is that the variable's value is getting changed too soon. The kind of output I'm expecting is something like this:

```
$C000: $0000 ; backlink to previous
$C002: "function name"

$C010: $C000 ; backlink to previous
$C012: "function2"

$C020: $C010; backlink to previous
$C022: "do-thing"
```

vhelin, 2020-02-05:
Yeah, but the assembler doesn't solve CADDR, the linker does. If the assembler could solve it it would work already...

MACRO  .term   ARGS    str
__\.\@:   .WORD   prev_term    ; < is writing CADDR-2 (from below), not the *previous* value
        .REDEF  prev_term __\.\£
        .BYTE   str.length, str, 0
.ENDM

?

Kroc, 2020-02-05:
`£` is not available on US keyboards, and isn't all that easy to type on a number of keyboard layouts; also it's quite an ugly hack :/ if there were a way to define a label from an expr, that might resolve the issue in a more broadly useful way.

vhelin, 2020-02-05:
If you define a label using that .LABEL, how do you create a reference to it? \< and \> are something I could do quickly and would mean \@-1 and \@+1. Hacks, yeah, but useful as well and can be done quickly, not next month. :)

Kroc, 2020-02-05:
It'll have to do otherwise it'll be impossible for me to continue programming without changing the assembler, again!

vhelin, 2020-02-06:
Ok, I'll add support for these today evening when I get back home...

vhelin, 2020-02-06:
Hmm, I might actually be able to make the \\@ parser to understand very simple additions and subtractions like \\@-1, \\@+9 and \\@+3... It'd be a parser hack and not very general, but at least it'd be easy to remember and use... Let's see!

Kroc, 2020-02-06:
That's a _lot_ nicer! Thank you so much for tackling this so quickly!

vhelin, 2020-02-06:
I've almost added the new feature, but now that I think of this case, doesn't this work for you?

```
.MACRO  .term   ARGS    str
__\.\@:   .WORD   prev_term
        .REDEF  prev_term __\.\@
        .BYTE   str.length, str, 0
.ENDM
```

?

vhelin, 2020-02-06:
```
.MACRO  .term   ARGS    str
__\.\@-1:   .WORD   prev_term
          .REDEF  prev_term __\.\@-1
          .BYTE   str.length, str, 0
.ENDM
```

works now, if you need it...

Kroc, 2020-02-06:
In all the confusion with `CADDR` (which had a bug as well), I didn't notice that this would work:

```
.MACRO  .term   ARGS    str
\.\@:   .WORD   prev_term
        .REDEF  prev_term       \.\@
        .BYTE   str.length, str, 0
.ENDM
```

The logic in your example won't work as the zeroth call cannot refer to a "-1" call

closed 2020-07-09 by vhelin