vhelin/wla-dx #291: WLA65816 doesn't put RAM labels in the Symbols file.

oziphantom, 2020-02-08:
say you declare some variables as follows
```
.ENUM mtmp1
ZPTemp1 db
ZPTemp2 db
ZPTemp3 db
OAMTemp ds 12
DESLogoTextNum db
DESLogoXBase dw
DESLogoYBase dw
DESLogoXDelta dw
DESLogoYDelta dw
DESLogoXAcc dw
DESLogoYAcc dw
OAMTempEditable ds 12
.ende
```
and you want to add some to watch window, or to a LUA file for you to be able to edit in the running emulator. 
One catch, you don't know where they are. They don't show up in the symbol file, the listing doesn't have an address or hex bytes in it so you can't look for an example of it being used to find the address. 

You have to add a fake label before where each one is used, boot the emulator, add breakpoints, reset the emulator, note down all the addresses and then go from there. Each time they change you have to repeat the process..

vhelin, 2020-02-08:
I tried both -S and -s, and these were present in the symbol file:

```
.ENUM $FFF0 EXPORT
scroll_x  DB
scroll_y  DB
player_x: DW
player_y: DW
.ENDE
```

If you don't use EXPORT then the labels are not visible outside that object file (to the linker)...

vhelin, 2020-02-08:
Also, for RAM labels, check out .RAMSECTION...

oziphantom, 2020-02-09:
ok EXPORT does the trick

closed 2020-02-09 by oziphantom