vhelin/wla-dx #292: Parameter substitution at start of label not working

Drenn1, 2020-02-15:
I have the following macro:

```
.macro m_soundPointer
	.db :\1Start - $39
	.dw \1
.endm
```

That is, the first parameter becomes part of a label name. So if the parameter is "sound39", this should translate to:

```
	.db :sound39Start - $39
	.dw sound39
```

Instead it gives an error:

```
build/audio.o: include/musicMacros.s:130: PARSE_STACK: Unresolved reference to "\1Start".
```

This used to work, but broke sometime between July and now. It seems to only be a problem at the start of a label (ie. something like "sound\1" works). I've attached a small project demonstrating the issue.
[macro_bug.zip](https://github.com/vhelin/wla-dx/files/4209105/macro_bug.zip)

vhelin, 2020-02-16:
I hope it works now...

Drenn1, 2020-02-16:
That seems to have fixed the issue, but like a game of whackamole, more regressions have come up in my project. I'll continue to investigate... thanks for the quick response :)

closed 2020-02-16 by Drenn1

vhelin, 2020-02-16:
With every whack I can add new tests to WLA DX's test suite so eventually this whole thing is just going to work. We just need to endure the whack-a-mole game... :)

vhelin, 2020-02-16:
And yes, the parser is horrible. :D Fix one thing and it might break another...

Drenn1, 2020-02-16:
About the test cases - I think I've mentioned this before, but could we add verification to them? Either as md5sum checks, or by having the expected result committed as well? There's the potential for silent failures if there isn't any verification on the example projects.

vhelin, 2020-02-16:
Every new test I write uses the byte_tester tool. With that we can check that the bytes the linker generate are what we expect them to be. If not, then run_unit_tests.sh will fail...

When running the script, every output like this:

   Test "TEST-01" SUCCEEDED!

is using byte_tester and tests that 1-N bytes in the output match.

Drenn1, 2020-02-16:
Aha, very good.