vhelin/wla-dx #293: Errors with getting label bank number

Drenn1, 2020-02-16:
I really have no idea what's going on here, so I'll just let the code speak for itself.

```
.include "cgb_hardware.i"

.MEMORYMAP
SLOTSIZE $4000
DEFAULTSLOT 1
SLOT 0 $0000
SLOT 1 $4000
SLOTSIZE $2000
SLOT 2 $c000
.ENDME

.ROMBANKSIZE $4000
.ROMBANKS 2

.emptyfill $ff

.ORGA $150

myLabel:
	.db $01 :myLabel ; Silent failure: only $01 is defined
	.db $01, :myLabel ; This works

	.db :$01 | :myLabel ; Gives the error below:
	; _GET_BANK_OF_ADDRESS: SLOT -1 < 0! Internal error. Please submit a bug report.
	; main.o: main.s:29: COMPUTE_STACK: Could not get the bank number for ROM address 1/$1 (out of bounds).

```

vhelin, 2020-02-16:
.db $01 :myLabel ; Silent failure: only $01 is defined

Due to a recent change, ':' is now a operation for the internal stack calculator, thus this transforms into a stack calculation of $01, OP_GET_BANK_NUMBER, myLabel -> as we parse through the stack there is nothing that would join the first $1 and OP_GET_BANK_NUMBER(myLabel), $1 is just what remains...

Drenn1, 2020-02-16:
Remember that parser rewrite that only supported commas for macro delimiters? The guy might have had a point =P

vhelin, 2020-02-16:
Duh... :D 

The error related to this one:

.db :$01 | :myLabel

is because a plain $01 has no SLOT associated to it. If it was :(myLabel+1) we'd know it's ROM bank... Without the SLOT association, perhaps I can just use .ROMBANKSIZE to calculate the BANK for this address?

Drenn1, 2020-02-16:
Interesting. I, uh... didn't mean to write ":$01", I meant to write "$01". Whether you consider that a bug or not is up to you...

This is the REAL bug:

```
	.db $01 | :myLabel ; Gives the error below:
; main.o: main.s:23: COMPUTE_STACK: OR is missing an operand.
```

vhelin, 2020-02-16:
That's a parser feature. 

.db $01 | (:myLabel)

should work.

vhelin, 2020-02-16:
Ok, I might have managed to fix the first bug in this report... Can you verify?

Drenn1, 2020-02-16:
Yeah it seems to be fixed. Still need to wrap a lot of bank references in brackets though.

vhelin, 2020-02-16:
That I cannot fix today, or very soon, it'll take a bigger effort (and reunderstanding of how the whole stack calculator thing works)...

vhelin, 2020-02-16:
Previously ':' was part of the label and had nothing to do with the stack calculator, and WLALINK could easily spot the ':' at the beginning of the label and calculate the bank, but now that ':' is a real operator in the stack calculator it'll go through a much bigger system than before...

Drenn1, 2020-02-16:
This is a precedence issue, right? The ":" operator needs to be handled before anything else. Effectively it should be treated as if it's wrapped in brackets anyway... easier said than done perhaps? I've never really understood the stack.

Drenn1, 2020-02-16:
Also, did the behaviour of this change in your last commit?

```
:label - $1a
```

It used to give me that COMPUTE_STACK internal error, which makes me suspect it was interpreted as ":(label - $1a)". But now it works as expected.

vhelin, 2020-02-16:
> The ":" operator needs to be handled before anything else. Effectively it should be treated as if it's wrapped in brackets anyway... 

Yes. And yes, easier said than done. :)

vhelin, 2020-02-16:
Even ":(label - $1a)" should work now. We have a test like this:

        .db ((:more_data)+1) + (1+(:(more_data+1))) + (more_data - $8000)

and it assembles just fine...

vhelin, 2020-02-17:
If you try now,

.db $01 | :myLabel

should work also! Once again...

vhelin, 2020-02-17:
It was actually a small and easy fix. I just didn't understand it yesterday. :D

Drenn1, 2020-02-20:
Sweet, glad it worked. :)

closed 2020-02-20 by Drenn1