vhelin/wla-dx #295: Error build on Mac from head

tsupplis, 2020-02-16:
Building now fails on Mac:

```
[  3%] Building C object CMakeFiles/wla-6809.dir/main.c.o
In file included from /Users/thierry/Projects/Devel/wla-dx/main.c:32:
/Users/xxxx/Projects/Devel/wla-dx/printf.h:91:9: error: 'snprintf' macro
      redefined [-Werror,-Wmacro-redefined]
#define snprintf  snprintf_
        ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.15.sdk/usr/include/secure/_stdio.h:56:9: note: 
      previous definition is here
#define snprintf(str, len, ...) \
        ^
In file included from /Users/thierry/Projects/Devel/wla-dx/main.c:32:
/Users/xxxx/Projects/Devel/wla-dx/printf.h:92:9: error: 'vsnprintf' macro
      redefined [-Werror,-Wmacro-redefined]
#define vsnprintf vsnprintf_
        ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.15.sdk/usr/include/secure/_stdio.h:74:9: note: 
      previous definition is here
#define vsnprintf(str, len, format, ap) \
        ^
2 errors generated.
make[2]: *** [CMakeFiles/wla-6809.dir/main.c.o] Error 1
make[1]: *** [CMakeFiles/wla-6809.dir/all] Error 2
make: *** [all] Error 2
````

vhelin, 2020-02-16:
> /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer
> /SDKs/MacOSX10.15.sdk/usr/include/secure/_stdio.h:56:9: note: 
>      previous definition is here
> #define snprintf(str, len, ...) \

I wonder, if we build this as an ANSI C90 project, why is snprintf() defined in the _stdio.h header file? It shouldn't be there? Or is the ANSI C90 -flag having no effect?

I have a MacBook at work, I could take a look at this there after work...

tsupplis, 2020-02-17:
I tried with -std=c90 without any luck. 
```
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc  -std=c99 -DMC6809 -DUNIX -I/Users/thierry/Projects/Devel/wla-dx  -pedantic-errors -Wall -Wextra -O2 -g -DNDEBUG -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.15.sdk   -o CMakeFiles/wla-6809.dir/main.c.o   -c /Users/thierry/Projects/Devel/wla-dx/main.c
In file included from /Users/thierry/Projects/Devel/wla-dx/main.c:32:
/Users/thierry/Projects/Devel/wla-dx/printf.h:91:9: error: 'snprintf' macro redefined [-Werror,-Wmacro-redefined]
#define snprintf  snprintf_
        ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.15.sdk/usr/include/secure/_stdio.h:56:9: note: 
      previous definition is here
#define snprintf(str, len, ...) \
```

tsupplis, 2020-02-17:
For information
```
cmake --version
cmake version 3.16.4

CMake suite maintained and supported by Kitware (kitware.com/cmake).
```
and Xcode is version 11.3.1

vhelin, 2020-02-17:
I hope my commit will fix building on macOS. At least I was able to get it to work with these changes... For some reason there already exists a definition of snprintf() and vsnprintf() in _stdio.h on Mac...

tsupplis, 2020-02-18:
Fixed. Build successful. thank you

closed 2020-02-18 by vhelin