vhelin/wla-dx #296: Allow skipping of bytes in RAM sections

Kroc, 2020-02-17:
On a console like the SMS, all RAM is always available for the programmer, but this is not always the case on microcomputers in certain situations; this affects the 6502 most because of the limited size of the zero-page.

There are only 256 bytes in the zero-page and all operations on the zero-page are faster than other memory address, so use of the zero-page is critical for speed. However, on a microcomputer, the existing OS/BIOS will already be using some of the zero-page space, and a programmer has to carefully work around zero-page addresses already in use. What addresses are used/available varies completely from one system to another

The typical way of doing this is just manually assigning addresses one-by-one as constants, but I'm missing out on using RAM structures because of this. One way of handling this may be to define a multitude of forced sections to block-out those bytes of RAM, but that requires  a massive amount of tedious up-front work that will put programmers off of doing this.

Whilst not a total solution for "mark these bytes as unusable within this SLOT" (perhaps a feature to consider), we can lighten the programmer's load by allowing the skipping of bytes within RAMSECTIONs (or even SECTIONs in general)

I can think of two ways, syntax wise to do this:
* allow for no label name, just a size (`DB`, `DW` &c) on a line to skip that many bytes
* a reusable, anonymous label name to indicate a skip, such as `.`

```
.RAMSECTION "zero_page" SLOT $00
    ; (without anonymous label)
            DB ; $00 is the C64's processor port data-direction-register
    ; (with anonymous label)
    .       DB ; $01 is the C64's processor port
    var     DB ; my zero-page variable
.ENDS

vhelin, 2020-02-17:
I'll try to make something like that, that works, in the next couple of days...

vhelin, 2020-02-18:
Byte(s) skipping using "." as a label sounded good. I hope it works!

Kroc, 2020-02-19:
Awesome, thank you so much!

closed 2020-02-19 by Kroc