vhelin/wla-dx #297: Support `FORCE` on `RAMSECTION`s

Kroc, 2020-02-19:
This is #296 pt.II; currently all `RAMSECTION`s behave as `FREE` sections. For microcomputers with reserved RAM addresses for BIOS/OS, we want to be able to define the system-dependent `RAMSECTION`s as FORCE (i.e. they use the `ORG` they were given), but the application-defined `RAMSECTION`s are free, and will automatically be placed in whatever space remains.

As a rough example, let's imagine that we have a program we're developing that needs 128 bytes of zero-page space to itself:

```
.RAMSECTION "myprogram" SLOT 0
    myvars    DSB 128 ;...
.RAMS
```

The program is designed to be compiled for more than one 6502-based system. One of these computer system uses the last 128 bytes of zero-page for itself:

```
.ORG $80
.RAMSECTION "system1" SLOT 0
    reserved    DSB 128
.ENDS
```

But system 2, happens to reserve the first 128 bytes of zero-page for itself:

```
.ORG $00
.RAMSECTION "system2" SLOT 0
    reserved    DSB 128
.ENDS
```

The desired output therefore is that for system1 `myvars = $00` and for system2, `myvars=$80`.
Assuming understanding of the documentation this would naturally be implied by using `FORCE` on `RAMSECTION`s, i.e.

```
.ORG $80
.RAMSECTION "system1" FORCE SLOT 0
    ; ...
```

vhelin, 2020-02-19:
I hope the latest commit fixes the issue.  I added ORG/ORGA support to .RAMSECTION as well.

Kroc, 2020-02-19:
That was amazingly fast! Is `ORG`/`ORGA` interchangeable in `RAMSECTION`? How does this impact the `ORG` of code that follows a `RAMSECTION`??

vhelin, 2020-02-19:
ORG/ORGA in .RAMSECTION should override .ORG and .ORGA... Just for convenience, and making sure you get what you want.

If you use ORG/ORGA in .RAMSECTION it'll alter the ORG/ORGA after the .RAMSECTION, I think.

vhelin, 2020-02-19:
Just double checked - it should be so. Too much beer - need to double check things. :D

vhelin, 2020-02-19:
You can use only ORG or ORGA in a .RAMSECTION, not both at the same time, or you can just not use them...

Kroc, 2020-02-19:
Thanks, appreciated! :)

Kroc, 2020-02-27:
I've been able to put this into practice (https://github.com/Kroc/pling/commit/6020e3721b033a6b6c85cdc323cac3b25e8130f8) and it works really well. I can now separate the zero-page addresses used by any 6502-based system, and the variables the program needs.

closed 2020-02-27 by Kroc

vhelin, 2020-02-27:
Glad to hear that the feature works and is useful! :)