vhelin/wla-dx #298: Error compiling SNES ROM: "MAIN: Error opening file ".wla4212a"."

dumuzidt, 2020-02-28:
I tried compiling an SNES .asm file with different versions of pre-built binaries, and they all give this error:

`MAIN: Error opening file ".wla[random numbers]a".`

Here is what I wrote in cmd for reference:

`wla-65816 -o first.o first.asm`

Let me know if there's any more info that would be helpful to solving this issue.

vhelin, 2020-02-28:
Are you using prebuilt Win32 executables from
http://www.villehelin.com/wla-win32.html ?

Weird. Do you have write access to the folder where you are running the exe
in?

vhelin, 2020-02-28:
I just downloaded the 9.10 executables, and ran this (using one example
from the 65816 folder):

D:\wla_dx_9.10>wla-65816.exe -o main.o main.s
Expected checksum: 3c6
Expected complement checksum: fc39

-> worked just fine.

dumuzidt, 2020-02-28:
Yes, I'm using the most up to date (9.2) win32 binaries from the site you linked, and I think there was an issue with Windows 10 permissions and the documents folder, so I moved everything to a different folder and now I'm getting a different error:
`INCLUDE_FILE: Error opening file "first.o".`

vhelin, 2020-02-28:
Can you run the assembler as admin? It should have write access to all
folders... I think this issue is because the user who runs the assembler
doesn't have write access to the current folder and cannot create files (a
temp file and an object file)...

vhelin, 2020-02-28:
> 
> 
> Yes, I'm using the most up to date (9.2) win32 binaries from the site you linked, and I think there was an issue with Windows 10 permissions and the documents folder, so I moved everything to a different folder and now I'm getting a different error:
> `INCLUDE_FILE: Error opening file "first.o".`

BTW, v9.2 is ancient. :) The most recent binaries on that page are for 9.10, and we are working on 9.11 currently...

dumuzidt, 2020-02-28:
Gotcha, I downloaded 9.2 because I'm so used to seeing the latest version at the top of the website haha. Just downloaded and tested 9.10 and it works, thanks for the help!

closed 2020-02-28 by dumuzidt