vhelin/wla-dx #300: undefined reference to `vnsprintf'

Drenn1, 2020-03-12:
Getting this error when trying to build on Linux, caused by commit 79111b90b86683bb246ec936aca5221d9b310290.

```/usr/bin/ld: CMakeFiles/wla-6502.dir/pass_1.c.o: in function `output_assembled_opcode':
pass_1.c:(.text+0x1b0d): undefined reference to `vnsprintf'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/wla-6502.dir/build.make:283: binaries/wla-6502] Error 1
make[2]: Leaving directory '/home/matthew/hdd/3rdprograms/wla-dx'
make[1]: *** [CMakeFiles/Makefile2:589: CMakeFiles/wla-6502.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
```

Drenn1, 2020-03-12:
Upon closer inspection, it seems that I accidentally (or intentionally but forgot) turned on the "DEBUG" flag at some point. Deleted CMakeLists and CMakeCache.txt to get rid of that. It compiles without the debug flag. I guess this should be fixed regardless though.

vhelin, 2020-03-14:
I hope it works now! The debug version at least compiles on my machine...

closed 2020-03-20 by vhelin

vince94, 2022-03-11:
Just encountered this while trying to build https://github.com/bazz1tv/snestracker natively on Windows via MSYS

vhelin, 2022-03-12:
> Just encountered this while trying to build https://github.com/bazz1tv/snestracker natively on Windows via MSYS

Not sure what you are saying. You were trying to build the latest WLA DX from sources? I mean if you are trying build that snestracker you need WLA executables... ?

vhelin, 2022-03-12:
> Just encountered this while trying to build https://github.com/bazz1tv/snestracker natively on Windows via MSYS

Now I see the issue:

WLA DX is a Git submodule inside that project, and the version of WLA it uses is 2 years old:

https://github.com/bazz1tv/wla-dx/tree/8e2c7986970a50228b84d3d4399f6f8573bca1d8

Please ask bazz1t to update snesntracker's WLA DX submodule to something more recent, so this is not a WLA DX issue but a snestracker issue.