vhelin/wla-dx #302: Allow `ORG`/`ORGA` in link files?

Kroc, 2020-03-25:
To help with assembling two versions of a piece of code -- one an original, unmodified fixed layout, and a second modern, unfixed layout I'd like to be able to have a link script that could place sections / ramsections at fixed addresses, so that this does not have to be done in the source files themselves.

For example, the RAM used for sound engine in Sonic1 is at $DC04, and whilst I can use `FORCE` to do this, I would like to also be able to assemble a version without this fixed address for when I make improvements to the game.

This could be facilitated by specifying a fixed address in the link script, e.g.

```
[ramsections]
BANK 0 SLOT 3 ORGA $DC04 "sound_RAM"
```

vhelin, 2020-04-01:
Currently only RAM sections can be touched in a linkfile using [ramsections] - should I add [sections] for normal code sections as well?

Kroc, 2020-04-01:
I don't know if that's horribly impractical to do, but yes, that would basically solve the problem of having to compile an original binary where every byte must be in the correct place, along-side a version where it doesn't matter where stuff goes and the linker can decide. I've been fighting with this issue for years and years across different assemblers!

vhelin, 2020-04-01:
I hope the latest commit fixes this issue. It wasn't that big thing to do, codewise... :)

Kroc, 2020-04-01:
I'll compile it now, but it'll take a while to put it through its paces; thanks!

Kroc, 2020-04-01:
Quick Question: Are Slot names supported in link files?

vhelin, 2020-04-01:
> Quick Question: Are Slot names supported in link files?

Yes they are. :)

Kroc, 2020-04-02:
I think there's a bug with how sections from libraries are handled. I moved my IRQs (`rst x`) into a library and used the linker script to `ORGA` them, but they are not being 'forced'; any section from anywhere else [that is not forced] is being given precedence. For example, a section from the main body of code is being placed first, and then an RST handler is being placed high up somewhere, like `$3D7A` (causing the link to fail because the address is no longer 8-bit)

vhelin, 2020-04-02:
> I think there's a bug with how sections from libraries are handled. I
> moved my IRQs (rst x) into a library and used the linker script to ORGA
> them, but they are not being 'forced'; any section from [...]

What's the type of these sections? Is the IRQs section a FREE section? FREE
and such sections float into suitable places, FORCE sections stay where
they are put into... Should the linkfile also be able to change the type of
the sections? Could this be the reason for the weird behavior?

Kroc, 2020-04-02:
The IRQ sections have no type because they are assembled into a library! https://github.com/Kroc/Sonic1-Z80-ASM/blob/redo/src/irqs.asm

Kroc, 2020-04-02:
Link script is here: https://github.com/Kroc/Sonic1-Z80-ASM/blob/redo/link_original.ini

vhelin, 2020-04-02:
> The IRQ sections have no type because they are assembled into a library!
> https://github.com/Kroc/Sonic1-Z80-ASM/blob/redo/src/irqs.asm

Ah, forgot that libraries don't take FORCE sections. But a section without
a type specifier is a FREE section thus it can move from the ORG/ORGA,
which is a starting point for it... If some other section is placed first
there then the following FREE sections move forward in memory...

What if you give the IRQ sections higher priority using PRIORITY?

Kroc, 2020-04-02:
Yes, that did work! I hadn't thought of that.

An ideal solution would be to allow `[sections]` in the link script to optionally specify `FREE`, `SEMIFREE`, `FORCE` etc -- if such a thing is possible -- so that we can have different placement behaviour between two different links using the same source code.

vhelin, 2020-04-02:
> Yes, that did work! I hadn't thought of that.
>
> An ideal solution would be to allow [sections] in the link script to
> optionally specify FREE, SEMIFREE, FORCE etc -- if such a thing is
> possible -- so [...]

Sounds like a good feature! I'll try to implement that later today, perhaps
even support for PRIORITY as well in the linkfile...

vhelin, 2020-05-13:
I guess this issue can now be closed?

Kroc, 2020-05-14:
Yes, absolutelyl this is a total game changer for me, so thank you!

closed 2020-05-14 by Kroc