vhelin/wla-dx #303: `.SMSHEADER` does not include `ROMSIZE` override

Kroc, 2020-03-25:
The `.SMSHEADER` directive does not include a 4-bit field, required to be able to duplicate existing ROMs:

[image]

WLA-DX *is* correctly calculating the last 4 bits according to the actual ROM size output, but this value is a guide to the BIOS for how to process the checksum -- it's value is not necessarily accurate, especially with 3rd-party mappers, which the BIOS does not support.

I'd like an optional `ROMSIZE` field for the `.SMSHEADER` directive, to manually override this value.
See https://www.smspower.org/Development/ROMHeader for ROM size specification

Kroc, 2020-03-26:
Addendum: The rom size is also incorrect when using `.BACKGROUND`; I'm assembling 64KB from source, and the rest (of a 256KB ROM) is being provided by the background, but WLA-DX is writing "$c" for the last nybble, to represent 32KB.

[image]

vhelin, 2020-04-03:
For some reason (the code was based on some old documentation?) WLALINK wrote the ROMSIZE automatically supporting only 8KB, 16KB and 32KB ROMSIZEs...

Kroc, 2020-04-05:
Bug with checksum calculation? 

[image]

(via this commit: https://github.com/Kroc/Sonic1-Z80-ASM/commit/43c4c9677f5a021a6f4dc1b42fbce3f97ceb9328)

Despite every other byte being 1:1 in the output, the checksum is incorrect.

vhelin, 2020-04-05:
So in the case of forced ROMSIZE 256KB ROM ($0), what is calculated to the checksum? Everything but not 0x7FF0-0x7FFF?

vhelin, 2020-04-05:
Ok, I made an assumtion that if the ROMSIZE is >32KB, then we'll checksum all the ROM except 0x7FF0-0x7FFF. Does this work?

Kroc, 2020-04-05:
Problem is fixed! You're a gentleman and a scholar; this was the last couple of bytes off of a perfect rebuild of Sonic 1 :)

closed 2020-04-06 by Kroc