vhelin/wla-dx #305: "You cannot include a file inside a MACRO". Why not?

Drenn1, 2020-04-26:
Maybe there's a good reason for this, but I thought I would open a discussion on it at least.

My project builds 2 different games; a flag toggles which one it builds, which in turn should affect which files it includes for certain purposes. I wanted to create a macro which would include a file, but decide which directory to include it from based on which game it's building. Something like this:

```
.macro include_build
	.ifdef ROM_AGES
		.define INCLUDE_BUILD_PARAM "build_ages_v/" \1
	.else
		.define INCLUDE_BUILD_PARAM "build_seasons_v/" \1
	.endif

	; No, we can't just ".include" it, we need to use a definition,
	; if we're doing a string append. (Separate bug report?)
	.include INCLUDE_BUILD_PARAM
	.undefine INCLUDE_BUILD_PARAM
.endm
```

This gives the following error:

```
build_ages_v/textDefines.sinclude/macros.s:126: DIRECTIVE_ERROR: You cannot include a file inside a MACRO.
include/macros.s:126: ERROR: Couldn't parse ".include".
```

There is no issue with using ".incbin" in a macro, so what's different?

This isn't "critical" or anything; up until now I've just been using symlinks to handle the issue. But I was curious what the reasoning behind this is.

vhelin, 2020-04-26:
It could be that at the time I implemented .MACRO for the first time, having and .INCLUDE inside a .MACRO was not going to work for some reason. I'll take a look at this next week, perhaps I can make it to work. :)

vhelin, 2020-04-26:
FYI, .INCBIN is just pure .DB in the end, .INCLUDE will change some pointers at least...

Drenn1, 2020-04-26:
That's a fair point, and I see you had to deal with .incbin as a special case. I guess this could get into slightly strange territory. Should the included file recognize the macro arguments (that is, be considered "inside" the macro)? ...Probably not, I think, considering the filename can't even be resolved until the macro is invoked.

vhelin, 2020-05-14:
This is a tricky situation. When you do a macro call we save the pointer to the place where we jumped from to the macro so that we can return to it when we encounter .ENDM. Now if macro is lexically before this jump point, and we include source code inside the macro, the source file will grow by the included bytes and the jumped-from-to-this-macro point gets put further away in the memory. So we'll need to increase the jumped-from-to-this-macro pointer by the bytes we just included after preprocessing, for all macro calls in the call stack.

I have this partially working now, but it might take a little bit more time to finish... And after my tests pass I'm still not 100% sure all will work. Perhaps time will tell us. :)

vhelin, 2020-05-14:
> That's a fair point, and I see you had to deal with .incbin as a special case. I guess this could get into slightly strange territory. Should the included file recognize the macro arguments (that [...]

In theory you are still inside the macro that contains the include, when you are parsing that include. Accessing the macro's arguments inside the include might work, but I'll let you know later after I do some tests. :)

vhelin, 2020-05-15:
It might work now, at least all my tests pass. And yes, if you .INCLUDE source.s inside a .MACRO, then you can make references to the .MACRO's arguments as source.s is inside that .MACRO...

Drenn1, 2020-05-16:
Appreciate your work on this. Though I'm encountering some entirely nonsensical errors when trying to use it. I don't really understand it but I've tweaked your example enough to reproduce similar errors. See: [example.zip](https://github.com/vhelin/wla-dx/files/4639119/example.zip)

I'm a bit unsure how I feel about the ability to use macro arguments inside an .include'd file, or the idea that it behaves like it's inside a macro in general. It wasn't what I had in mind, but I suppose this is probably the least-surprising way to implement such a feature.

vhelin, 2020-05-16:
Ouch, thanks for the feedback! I'll take a look at tomorrow why included .MACROs don't seem to work...

vhelin, 2020-05-17:
> Appreciate your work on this. Though I'm encountering some entirely nonsensical errors when trying to use it. I don't really understand it but I've tweaked your example enough to reproduce similar errors. See: [example.zip](https://github.com/vhelin/wla-dx/files/4639119/example.zip)

I forgot to apply the same offsetting to a (non-running) .MACRO's starting point in memory that I applied to currently running .MACROs, now it should work better...

> I'm a bit unsure how I feel about the ability to use macro arguments inside an .include'd file, or the idea that it behaves like it's inside a macro in general. It wasn't what I had in mind, but I suppose this is probably the least-surprising way to implement such a feature.

It might also feel a bit weird to stop running a .MACRO when an .INCLUDE is encountered, and then resume the .MACRO after the included file ends... That would be something you didn't expect when looking at the source code...

vhelin, 2020-05-31:
Any news about this, is it working 100% for you? Can we close this issue?

Drenn1, 2020-06-06:
Using the same macro to include 2 different files appears to cause the same file to be included twice. If you can't reproduce the problem I'll try to make another test case.

vhelin, 2020-06-07:
This works for me:

.MACRO MACRO_INCLUDE_2
        .INCLUDE "include3.s"
        .INCLUDE "include4.s"
.ENDM

both are included only once when I call MACRO_INCLUDE_2... So I need your test case to proceed. :)

Drenn1, 2020-06-07:
I didn't explain very well - I mean when you use a parameter to include a file, the first invocation of the macro seems to affect the second invocation. I've attached my test case.
[testcase.zip](https://github.com/vhelin/wla-dx/files/4742317/testcase.zip)

vhelin, 2020-06-07:
Yeah, well, this is a tricky case.

The first time you call INCLUDE_MACRO "include1.s" the file "include1.s" is loaded and added to INCLUDE_MACRO. The next time you call INCLUDE_MACRO "include2.s" it sees that the .INCLUDE inside the macro has been used already and skips it, and parses the contents of "include1.s" again as that's been included into INCLUDE_MACRO.

If I disabled .INCLUDE disabling inside a .MACRO we'd parse "include2.s" and "include1.s" inside the second INCLUDE_MACRO call - you'd not want this (case 1). If I skipped the included "include1.s" inside the second call of INCLUDE_MACRO - it wouldn't work as it should (case 2). Both cases would fail...

I guess the only logical outcome of this is that .INCLUDE \1 does not work. Unless it's ok that case 1 works as it follows the design or the parser...

Drenn1, 2020-06-07:
I originally opened this issue because I wanted to include a "variable name" in a macro, but after some time thinking about it, I think other solutions would work better for my use case. Another idea I had is that we could simply allow "string append" operations to work with the .INCLUDE directive, then I wouldn't need to deal with macros for this. I think this solution would make more sense for my use-case.

So, obviously it's up to you to decide which cases you want to support here. It's worth also considering what would happen if you use other .DEFINE'd values in a macro though. I think under normal circumstances the value wouldn't be resolved until the macro is invoked? In this case it would essentially cause the same problem as using a parameter.

vhelin, 2020-06-07:
> Another idea I had is that we could simply allow "string append" operations to work with the 
> .INCLUDE directive, then I wouldn't need to deal with macros for this.

Can you give an example of this, how it should work?

Drenn1, 2020-06-07:
```
.DEFINE INCLUDE_FOLDER "build_ages/"

.INCLUDE INCLUDE_FOLDER "data/fileToInclude.s"
```

Essentially it would work in the same way that string appends work for .DEFINE's right now. This works already if you create a 2nd define with the concatenated strings first:

```
.DEFINE INCLUDE_FOLDER "build_ages/"

.DEFINE INCLUDE_FILE INCLUDE_FOLDER "data/fileToInclude.s"

.INCLUDE INCLUDE_FILE ; Include "build_ages/data/fileToInclude.s"
```

 At least, it's my observation that putting two adjacent strings together concatenates them, but I'm not sure if that's in the documentation.

vhelin, 2020-06-07:
That's a good observation, I don't remember how that should work. :) If that's a concatenation then let's leave it to be that...

But adding support for such concatenation in .INCLUDE is very much doable...

vhelin, 2021-03-19:
What's the current status of this issue? Did the last commits fix it or is there still something that needs fixing?

vhelin, 2021-04-29:
Looking through the comments here I guess this issue can now be closed as .INCLUDE now works inside a .MACRO (as well as it can work, I think) and then we implemented some additional features here.

closed 2021-04-29 by vhelin