vhelin/wla-dx #307: wla-gb output files improperly link in program mode

karmic64, 2020-05-11:
When wlalink is used to link a wla-gb object file, padding is added regardless of whether or not the -b switch is used.

For example:
```
.memorymap
defaultslot 0
slot 0 $2000 $2000
.endme

.rombankmap
bankstotal 2
banksize $2000
banks 2
.endro

.bank 0 slot 0
.orga $3000
here: jp here
```

Assembling this source file with wla-z80 and linking it with -b works properly- the contents of the file is simply `C3 00 30`, but assembling it with wla-gb will insert $EB8 zero bytes at the start of the file when linked.

vhelin, 2020-05-11:
These lines in pass_2.c of WLA write a byte to $148 and the program size calculator uses it as the starting point for the program...

    if (rombanks_defined != 0) {
      /* create a what-we-are-doing message for mem_insert*() warnings/errors */
      snprintf(mem_insert_action, sizeof(mem_insert_action), "Writing GB ROM banks");

      mem_insert_absolute(328, romtype);
    }

Moving all those bits to WLALINK would take some time, but how about I add options to WLALINK that can be used to specify the starting and end addresses for the output program file?

vhelin, 2021-05-03:
-bS and -bE should fix this issue. If not, please reopen it.

closed 2021-05-03 by vhelin