vhelin/wla-dx #309: Macros for assembly time math

shewitt-au, 2020-05-13:
It seems macros can't be used in this fashion.

```
Start:
    .MACRO RGB ARGS R,G,B
        r|g<<4|b<<4
    .ENDM
    ldx #RGB 1 2 3
```

Somewhere in the documentation I do remember reading something about macros being more like subroutine calls so I fear that may be by design. If so is their some mechanism to achieve this?

vhelin, 2020-05-13:
Yeah, MARCOs in WLA are much like subroutines that don't return anything. Only this works:

.MACRO RGB ARGS R,G,B
    ldx #R|G<<4|B<<4
.ENDM
    
    RGB 1 2 3

vhelin, 2020-05-13:
This might also work, but is not worth the effort?

```
.MACRO RGB ARGS R,G,B
    .REDEFINE RGB_VALUE R|G<<4|B<<4
.ENDM
    
    RGB 1 2 3
    ldx #RGB_VALUE
```

:)

shewitt-au, 2020-05-13:
I've been impressed by wla-dx, but that's a real kick in the pants.

vhelin, 2020-05-13:
It might be, perhaps, possible to make the following to work:

```
.MACRO RGB ARGS R,G,B
    .RETURN R|G<<4|B<<4
.ENDM
    
    ldx #RGB 1 2 3
```

as

```
.MACRO RGB ARGS R,G,B
    R|G<<4|B<<4
.ENDM

    ldx #RGB 1 2 3
```

doesn't make sense and is way harder to parse. Perhaps I'll have the time and energy to make this work one day. Or any volunteers? :)

Kroc, 2020-05-14:
I too would love to see expression macro support in WLA-DX.

CA65 uses C-style `.define`'s which act as a straight in-line fragment replacement, but that's only possible because CA65 is essentially a C-compiler.

Perhaps allow the definition of custom "expressions", or use the C64's `DEF FN` syntax:

    .EXPR half_x ARGS x = x / 2
    .DEFFN half_x ARGS x = x / 2

Drenn1, 2020-05-14:
I think special thought needs to be given to how macro arguments are interpreted, in case you want to do this multiple times, ie. for something like this:

```
.macro ldbc
    ld bc, (\1<<8) | \1
.endm

.macro MULT_BY_2
   .RETURN \1 * 2
.endm

    ldbc MULT_BY_2 5, MULT_BY_2 6
```

Dumb example but you can see the point. The macro needs to understand where its arguments end and where "other things" resume. Or at least that would be preferable for maximum flexibility, for cases like this, or perhaps other cases such as nesting macro calls. I don't know if this is feasible, but I guess it would be nice if they used brackets?

```
   ld b, MULT_BY_2(MULT_BY_2(5))
```

vhelin, 2020-05-15:
All the options presented in this thread require some detective work and big or small changes to the parser(s) and how the rest of the code uses them - not an easy task... The .RETURN one might be the easiest one to code. Let's see...

Kroc, 2020-05-28:
90% of the plumbing is present through the filter-macro (`_OUT`) support, no?
E.g.

```
.MACRO .add
        .REDEF  _OUT    \1 + \2 
.ENDM
```

nickgirardo, 2021-05-31:
I know this issue is a bit dead, but has this been given any further consideration?  I came across this issue when looking for a better way to write a macro which included code similar to
```
; Store the low word part of the given address
ldx #(CPU_ADDR & $00FFFF)
stx $4302
; Store the bank part of the given address
lda #(CPU_ADDR >> 16)
sta $4304
```

While this code works, it isn't really clear exactly what it's doing on first glance and basically the same clunky arithmetic expressions have to be written across a handful of macros. 

CA65 (mentioned above) has [.loword](https://cc65.github.io/doc/ca65.html#ss10.15) and [.bankbyte](https://cc65.github.io/doc/ca65.html#ss10.3) directives for this purpose.  In lieu of modifying the macro system to enable expressions these directives would at least aide me for my use cases.  However, I also think it's clear that there are many other possibilities which could come from more general macros

vhelin, 2021-05-31:
I think that adding support for .MACROs that return values, or quick and small .MACRO like user definable functions requires quite a lot of work and might not be something that I can do very soon, but adding fixed/hardcoded functions like 

  loword()
  bankbyte()

is something I can do pretty quickly. We already have such functions like asc(), defined() and exists(). Would this be ok?

nickgirardo, 2021-05-31:
I do think that would be helpful, although long term it isn't really viable to have you implement every user-requested fn in the compiler :P

But for what is basically byte swindling I think that makes a lot of sense as an in-built feature of a macro assembler

vhelin, 2021-05-31:
I'm thinking that I'll quickly add loword(), hiword(), hibyte(), lobyte() and bankbyte() to WLA DX and release WLA DX v10.0 in 3-4 weeks. For v10.1 later when I have more time (and something like a summer vacation) I'm thinking that I could give it a try to make .MACRO calling using

  .macro macroname
  ...
  .endm

  macroname(arg1, arg2...)

to work.

nickgirardo, 2021-05-31:
Yeah, you could also consider having the directive name `.macroexpr ... .endm` if that would make disambiguation easier.  I think that might help with reading assembly as well as you would be able to more quickly tell if a macro were meant to expand into assembly or a value.

nickgirardo, 2021-05-31:
At least to me assembly generating macros (what is currently supported) and expression/ value producing macros (what this issue requests) feel very different conceptually so different directive names might help

vhelin, 2021-05-31:
Good point. We could also then make it a requirement that all .MACROEXPR macros need to return a value, always.

Then again it might be easier just to issue an error if the .MACRO macroname in macroname(arg1, arg2) doesn't return a value... I'll think more about this when I'm going to try to code this feature. :)

vhelin, 2021-05-31:
I'm currently implementing the functions, but need to go to sleep (00:57 here). I hope to commit them to the repo tomorrow evening.

nickgirardo, 2021-05-31:
Sounds great, thanks for being so responsive!

vhelin, 2021-06-01:
Note that I added also bank() which is one you'll want to use in your example:

; Store the bank part of the given address
lda #(CPU_ADDR >> 16)
sta $4304

->

lda #bank(CPU_ADDR)
sta $4304

or you could use the ':' operator:

lda #:CPU_ADDR
sta $4304

nickgirardo, 2021-06-01:
Thanks for taking care of this so quickly, I'm going to update wla-dx and check it out

nickgirardo, 2021-06-01:
It's working for me.  It was actually the `bankbyte` function that did what I needed, I'm not 100% sure how it differs from `bank`

vhelin, 2021-06-02:
bank() gives you the bank of the label (or pure ROM address value) (plus .BASE on 65816). bankbyte() gives you bits 16-23 of a value.

WLA doesn't have bank number (+ base) in a label's address value, but keeps bank number (and .BASE) in another variable. Thus only bank() or ':' can give you that value.

vhelin, 2021-06-02:
Hmm, wait a second, I'm looking at WLALINK's code and it seems that on 65816 the label's address contains bank + base in bits 16-23, but not always... ? I need to look deeper into this, it seems I've forgotten how stuff works on 65816. But on other systems you need to use bank() to get the label's bank number... :)

vhelin, 2021-06-02:
On, now one can use bankbyte() to get the bank (+ base) of an 65816 address label, even if the calculation result is not 24-bits. Which was a weird limitation and still is for some other operations.

65816:
Raw address value - use bankbyte() 
Address label - use bankbyte() though bank() works as well

Other CPUs:
Raw address value - use bank()
Address label - use bank() though bankbyte() works as well

nickgirardo, 2021-06-07:
Thanks for the update, I'll pull in the latest changes and check it out!

vhelin, 2022-03-12:
This is kinda duplicate of issue #7 here, so I'll move @Kroc's comment there as reminder of how this could be solved, and close this issue

closed 2022-03-12 by vhelin