vhelin/wla-dx #310: Macros only execute if indented

lajohnston, 2020-05-13:
Hi there,

I want to run a macro at the top level of a file but it didn't seem to be working. After trial and error it turns out the macro only executes if the call is indented by at least one space, i.e:

```
.macro "foo"
    .print "Bar\n"
.endm

 foo  ; this runs, note the space at start of line
foo   ; this won't run
````

Is this a known behaviour?

vhelin, 2020-05-13:
Yes, well, all strings that are placed at the beginning of the line, and are not directives, are considered as labels. It's been there for quite a while.

But I guess we could make macro calls work in this case. Though it might be a bit confusing for those who live by the old rule...

Kroc, 2020-05-14:
Most assemblers operate with the notion of a "gutter"; text that is _not_ indented is a definition, typically a label. Text that _is_ indented is the content, typically the assembly syntax. Some assemblers require there be a colon to define a label, but some rely on the guttering to determine what is and isn't a label, ala, the issue you're running into here.

lajohnston, 2020-05-14:
Ah interesting, ok I guess it might muddy the waters if it had a special exception for this, not without macros having their own syntax like `foo()` at least. The indentation's not the end of the world and I can keep it in with a comment to remind me. Thanks for clarifying!

vhelin, 2020-05-15:
This is actually a duplicate of issue #85 - https://github.com/vhelin/wla-dx/issues/85

Perhaps it might be worth making macro calls to work without leading whitespace as this old requirement seems to cause confusion...

closed 2020-05-31 by vhelin