vhelin/wla-dx #311: Add ALIGN to .RAMSECTIONs

vhelin, 2020-05-16:
Currently there is no way of aligning the .RAMSECTIONs...

Drenn1, 2020-05-16:
To clarify, we can ALIGN ramsections, but I was thinking of an .ALIGN directive which could work _within_ ramsections (and normal sections also). I had an idea of how this might be implemented in #173 which would not involve the linker at all (it would depend on assuming the entire section is aligned in a certain way), but it's possible I may have overlooked something.

Kroc, 2020-05-17:
Whilst not exactly a replacement for `.ALIGN`, you are currently able to skip bytes in RAM sections using the anonymous label `.`, e.g.

```
.RAMSECTION "something" ...
    one     .DB
    .       .DB
    two     .DB
.ENDS

vhelin, 2020-05-17:
Haha, I was a bit sleepy last night, should have checked if .RAMSECTIONs already had ALIGN... :)

vhelin, 2020-05-17:
> To clarify, we can ALIGN ramsections, but I was thinking of an .ALIGN directive which could work _within_ ramsections (and normal sections also). I had an idea of how this might be implemented in [...]

I re-read #173 and yes, if the assembler did the .ALIGN by requiring the .SECTION/.RAMSECTION itself to be aligned by something bigger, that would be relatively easy to do. But that would mean we'd only support special cases and users would most definitely report issues that they want .ALIGN to work fully no matter how the .SECTION/.RAMSECTION is aligned...

Making a universal support for .ALIGN, that would mean we'd need to leave it to the linker, and that'd be a pretty big job in the current framework, I think... It'd be a good feature to have, but I don't think I can do it anytime soon...

Drenn1, 2020-05-17:
Absolutely, it would be tricky. The size of a section would be unknown until it is placed. Not a can of worms I'd want to deal with.

I personally think the limited version would still be quite useful, and when people inevitably want it to be more flexible, you can put your foot down and close with "WONTFIX" :) Your call though.

> Whilst not exactly a replacement for `.ALIGN`, you are currently able to skip bytes in RAM sections using the anonymous label `.`, e.g.
> 
> ```
> .RAMSECTION "something" ...
>     one     .DB
>     .       .DB
>     two     .DB
> .ENDS
> ```

Good to know, I did not realize "." acted as an anonymous label. This solution does leave a bit to be desired of course.

vhelin, 2020-05-17:
> I personally think the limited version would still be quite useful, and when people inevitably want it to be more flexible, you can put your foot down and close with "WONTFIX" :) Your call though.

Uh, if I have to do something, I'd like to do it properly and so that no-one has any complaints about it afterwards. :) So linker will do it, and so that it'll work in all cases... But this needs planning and thinking.

vhelin, 2023-01-01:
Good news: I lowered my personal goals, and thanks to WLA's recent ability to calculate A-B deltas I can add .ALIGN that'll work in some cases. Commit coming soon...

vhelin, 2023-01-01:
> To clarify, we can ALIGN ramsections, but I was thinking of an .ALIGN directive which could work _within_ ramsections (and normal sections also). I had an idea of how this might be implemented in [...]

I need to go through this with a cup of coffee later this week. And also try to add that .ALIGN to .RAMSECTIONs as well...

vhelin, 2023-04-29:
I think .ALIGN is as good as it can get with the current architecture of WLA DX so I'm closing this issue.

closed 2023-04-29 by vhelin