vhelin/wla-dx #313: APPENDTO chains of 3 or more must be defined in a specific order

Drenn1, 2020-05-16:
Noticed while doing #312 that my appendto chains needed to be defined in a specific order. For example, this works:

```
bank 0 slot 2 orga $cc00 "Wram0_cc00"
bank 0 slot 2 appendto "Wram0_cc00" "Wram0_FloatingSection2"
bank 0 slot 2 appendto "Wram0_FloatingSection2" "Wram0_FloatingSection1"
```

But swapping the last 2 lines doesn't work:

```
bank 0 slot 2 orga $cc00 "Wram0_cc00"
bank 0 slot 2 appendto "Wram0_FloatingSection2" "Wram0_FloatingSection1"
bank 0 slot 2 appendto "Wram0_cc00" "Wram0_FloatingSection2"

```

```
MERGE_SECTIONS: Target section "Wram0_FloatingSection2" was not found, ignoring the "Wram0_FloatingSection1" -> "Wram0_FloatingSection2" append.
```

The cause is pretty clear; when "Wram0_FloatingSection2" is appended to "Wram0_cc00", then the section named "Wram0_FloatingSection2" disappears, and can no longer be appended to. So the appends must be declared in a specific order.

vhelin, 2021-03-18:
I rewrote your example the best I could in WLA syntax:

.bank 0 slot 2
.orga $cc00

.section "Wram0_cc00"
.ends

.section "Wram0_FloatingSection2" appendto "Wram0_cc00"
.ends

.section "Wram0_FloatingSection1" appendto "Wram0_FloatingSection2" 
.ends

...

What you are saying is true, but if you take now the latest WLA DX sources, there is a workaround:

.section "Wram0_cc00" PRIORITY 100
.ends

.section "Wram0_FloatingSection2" appendto "Wram0_cc00" PRIORITY 99
.ends

.section "Wram0_FloatingSection1" appendto "Wram0_cc00" PRIORITY 98
.ends

vhelin, 2021-05-01:
Please try out the latest sources, they allow now chaining of APPENDTO sections...

Drenn1, 2021-05-03:
I'll get back to you. Could you take a look at gb-z80? It seems totally broken on the latest commit, tests are failing right now.

vhelin, 2021-05-04:
All tests work on Linux, Cygwin and Windows... ?

vhelin, 2021-05-04:
Have you tried to issue "make clean"? As I've modified opcodes_gb.c you'll need to do "make clean" to get working executables...

closed 2021-05-25 by vhelin