vhelin/wla-dx #314: Muting warnings for cross-stack calculations

Drenn1, 2020-05-16:
As discussed in #173, doing calculations involving labels from different slots causes a warning to be printed:

```build/seasons.o: code/bank0.s:1821: COMPUTE_STACK: The passed on SLOT changed from $0 to $2. This might have no effect, but just to let you know. Please check that the result of this calculation is correct.```

The warning makes sense and is useful. But there are legitimate reasons to need to do this, and having this warning printed every time my code is assembled clutters the console output. Could we have an option to mute this? Like "-Wno-slot-calc-check" in wlalink or something like that?

vhelin, 2020-05-17:
The reason I made WLALINK to give that error is that I'm not 100% sure that, in the case that happens, 100% of the calculations will be correct. :) They might be, but I'm just not sure myself...

Adding a switch to silence such warnings is pretty simple, though.

vhelin, 2022-02-05:
I think I removed these warnings so this issue should be closed... :)

closed 2022-02-05 by vhelin