vhelin/wla-dx #315: GameBoy ROMs should be padded

Kroc, 2020-05-18:
[image]

Using the following code:

```asm
.MEMORYMAP
        DEFAULTSLOT     0
        SLOT 0  START $0000 SIZE $4000 NAME "ROM-0"     ; 16KB ROM Bank 0
        SLOT 1  START $4000 SIZE $4000 NAME "ROM-1+"    ; 16KB ROM Bank 1-N
        SLOT 2  START $8000 SIZE $2000 NAME "VRAM"      ; 8KB VRAM
        SLOT 3  START $A000 SIZE $2000 NAME "ERAM"      ; 8KB External RAM
        SLOT 4  START $C000 SIZE $1000 NAME "WRAM-0"    ; 4KB WRAM Bank 0
        SLOT 5  START $D000 SIZE $1000 NAME "WRAM-1+"   ; 4KB WRAM Bank 1-N
.ENDME

.ROMBANKSIZE $4000
.ROMBANKS 2

.NINTENDOLOGO                   ; insert required Nintendo logo at $104-$133
.NAME   "BECAUSE"               ; game title (in ROM header), at $134-$143

.LICENSEECODENEW        "CD"    ; two-digit manufacturer code at $144-$145
                                ; (required for Super GameBoy compatibility)
                                ; BTW "CD" is for "Camen Design"

.ROMDMG                         ; Super GameBoy flag at $146; DMG or SGB

.CARTRIDGETYPE	        $03     ; select MBC1+RAM+Battery

.RAMSIZE	        $01     ; 2KB SRAM

.DESTINATIONCODE        $01     ; non-Japanese

.VERSION                $00     ; "ROM version"

.COMPUTEGBCOMPLEMENTCHECK       ; checksum byte at $14D
.COMPUTEGBCHECKSUM              ; checksum bytes at $14E/F

.EMPTYFILL	        $00

.BANK   0       SLOT 0
.ORG    $100

start:
        jp start
```

The BGB emulator will give this warning or fail with a similar error regarding the ROM file not being the expected size.

The `.ROMBANKSIZE` / `.ROMBANKS` indicates a 32 KB ROM; am I forgetting how to automatically pad the file or something?

vhelin, 2020-05-18:
You selected cartridge type to be MBC1+RAM+Battery, but 32KBs ROMs don't seem to have banking.

  https://gbdev.gg8.se/wiki/articles/The_Cartridge_Header#0148_-_ROM_Size

Should your cartridge type be $00? Perhaps that's the reason...

Kroc, 2020-05-18:
A cartridge type of $00 still produces the same error, the output file is 1 KB regardless

vhelin, 2020-05-18:
1KBs? I get 32KBs here...

vhelin, 2020-05-18:
$ ../binaries/wla-gb.exe -v -o main.o main.s
$ ../binaries/wlalink.exe -v linkfile output.rom

vhelin, 2020-05-18:
Are you making WLALINK to output a program file instead of a ROM file?

vhelin, 2020-05-18:
$ ../binaries/wlalink.exe -v linkfile output.rom
Free space at $0000-$00ff.
Free space at $0103.
Free space at $0150-$7fff.
Bank 00 has 16305 bytes (99.52%) free.
Bank 01 has 16384 bytes (100.0%) free.
32689 unused bytes of total 32768.

vhelin, 2020-05-18:
$ li
total 54
drwxrwxr-x+ 1 ville ville     0 May 18 21:15 ./
drwxrwx---+ 1 ville ville     0 May 18 21:13 ../
-rwxrw-r--+ 1 ville ville    17 May 18 21:14 linkfile*
-rwxrw-r--+ 1 ville ville   296 May 18 21:14 main.o*
-rwxrw-r--+ 1 ville ville  1357 May 18 21:13 main.s*
-rwxrw-r--+ 1 ville ville 32768 May 18 21:15 output.rom*

Kroc, 2020-05-18:
Curious, from `wlalink -v -b ...`, I get: 
```
Program start $0, end $14f.
Free space at $005e-$00ff.
Free space at $0150-$7fff.
Bank 00 has 16210 bytes (98.94%) free.
Bank 01 has 16384 bytes (100.0%) free.
162 unused bytes (48.21%) of total 336.
```

vhelin, 2020-05-18:
-b  Program file output :D

vhelin, 2020-05-18:
If you want a ROM file, don't use -b

Kroc, 2020-05-18:
Ah, so silly of me -- I copy and pasted my batch script from my C64 project! I'm sorry -- totally my fault!

closed 2020-05-18 by Kroc

vhelin, 2020-05-18:
Haha, no problem :)