vhelin/wla-dx #316: ".REPEAT needs a count" when using labels for the repeat count

hisahi, 2020-05-21:
On wla-65816 at least, it seems that `.REPEAT` (or `.REPT`) does not support having labels in the count parameter at all (when dealing with arithmetic), as doing so makes it think that no count parameter has been supplied.

Example:

```
TESTSTRING:
    .DB "test"
TESTSTRINGEND:
.REPEAT 16-_sizeof_TESTSTRING
    .DB " "
.ENDR
```

or, equivalently:
```
TESTSTRINGSTART:
    .DB "test"
TESTSTRINGEND:
.REPEAT 16+TESTSTRINGSTART-TESTSTRINGEND
    .DB " "
.ENDR
```

Both result in the error `INPUT_ERROR: .REPEAT needs a count` (during pass 1) when assembling instead of repeating the space character to pad the string to 16 characters. The same happens if I try enclosing the calculation in parentheses, like `.REPEAT (16+TESTSTRINGSTART-TESTSTRINGEND)`.

Is this a bug or simply a limitation? I could probably work around it in the latter case.

vhelin, 2020-05-21:
This is a limitation. WLALINK is the one that places the labels and can calculate TESTSTRINGSTART-TESTSTRINGEND, and WLALINK is also the one that generates those _sizeof_ -labels... In many ways the assembler is not very smart. :)

.REPEAT needs immediate count value in the very first pass...

hisahi, 2020-05-21:
That is understandable. I also tried using `.length` (this is for a macro so that it can pad a string passed as a parameter), but that resulted in the same error. I'll see if I can come up with a workaround.

vhelin, 2020-05-21:
The very first pass of the assembler should be able to calculate string.length... What kind of a .MACRO are you using?

hisahi, 2020-05-21:
Actually, it's very possible I was getting the error for something else, because this indeed seems to work perfectly:

```
.MACRO PADSTRING ARGS STRING, LENGTH, ERRMSG
.IF LENGTH < STRING.length
.PRINTT ERRMSG
.PRINTT "\n"
.FAIL
.ENDIF
.DB STRING
.REPEAT LENGTH-STRING.length
.DB " "
.ENDR
.ENDM
```

vhelin, 2020-05-21:
I tried this:

´´´
.MACRO DEFINE_BYTES
    .DB \1
.REPEAT 16-\1.length
    .DB " "
.ENDR
.ENDM

.section "!DUMMY4" FREE
        DEFINE_BYTES "hello"
.ends
´´´

... which is perhaps what you are not doing, but I'm getting an error for this:

main.s:120: INPUT_ERROR: .REPEAT needs a count.
main.s:120: ERROR: Couldn't parse ".REPEAT".
make: *** [makefile:15: main.o] Error 1

... which makes me think that the stack calculator in the assembler cannot solve \1.length for some reason...

Anyway, if you got that working - good! But I'll still look into the problem case I have here... :)

vhelin, 2021-08-01:
This issue is actually the same as #15 and #316. My latest commit https://github.com/vhelin/wla-dx/commit/3f5dd1cd020f484da308f55e9e711d9201d242a8 fixes this and the other issues as well. If you think it's not working, please reopen this issue.

closed 2021-08-01 by vhelin