vhelin/wla-dx #318: Allow `[C]` style instruction syntax for GameBoy

Kroc, 2020-05-31:
The Z80 has I/O ports which can be accessed with:

```
ld A, [$nn] ; read from port $nn
ld [$nn], A ; write to port $nn
ld A, [C]   ; read from the port-number given in register-C
ld [C], A   ; write to the port-number given in register-C
```
On the GameBoy, the I/O ports are mapped to some I/O & fast-RAM embedded into the CPU.
These are also exposed to the memory map:

```
$FF00-$FF7F = HW I/O
$FF80-$FFFE = embedded CPU "High RAM"
$FFFF       = Interrupt register
```

Whilst you can read/write directly to these memory addresses like any other memory address, access is faster / easier when done through the I/O instructions.

Because of the $FFxx address range, these instructions are often written in the form:

```
ld A, [$FF00+$nn]
ld [$FF00+$nn], A
ld A, [$FF00+C]
ld [$FF00+C], A
```

(I believe this syntax comes directly from Nintendo, who had to purposefully use non-Intel syntax to avoid copyrights)

This is the syntax WLA-DX currently supports. Whilst this syntax is perhaps more obvious of its intended effect, it is incompatible with Z80 syntax and that means that it's more difficult to share code between Z80 & GB, e.g. in projects that compile the same source to different targets.

Could you please add the Z80-syntax to GB too:

```
ld A, [$nn] ; = ld A, [$FF00+$nn]
ld [$nn], A ; = ld [$FF00+$nn], A
ld A, [C]   ; = ld A, [$FF00+C]
ld [C], A   ; = ld [$FF00+C], A
```

vhelin, 2020-05-31:
There is already support for (you can use '[' and ']' instead of '(' and ')')

  { "LDH A,(C)", 0xf2, 0, 0 },
  { "LDH A,(x)", 0xf0, 1, 0 }, // x is an 8-bit value
  { "LDH (C),A", 0xe2, 0, 0 },
  { "LDH (x),A", 0xe0, 1, 0 }, // x is an 8-bit value

... which should do the trick. And there are already

  { "LD A,(?)", 0xfa, 2, 0 }, // ? is an 16-bit value
  { "LD (?),A", 0xea, 2, 0 }, // ? is an 16-bit value

... so I cannot add support for

  ld A, [$nn] ; = ld A, [$FF00+$nn]
  ld [$nn], A ; = ld [$FF00+$nn], A

as they are doing different things while lexically they'd conflict.

Kroc, 2020-06-01:
My frazzled brain not noticing things again!
Perhaps add an error message for GB80 to suggest "did you mean 'ld**h** ....'"

closed 2020-06-12 by Kroc