vhelin/wla-dx #322: gpl3 compatibility?

bazzinotti, 2020-06-13:
Hi it's me again, Bazz. Sorry to make such a ruckus on this complacent issues tracker.

A project I recently licensed under GPL3 is facing "license woe" syndrome because of uncertainty with regards to wla-dx's license. https://github.com/bazzinotti/snestracker/issues/59

Is wla-dx GPL3 compatible? If not, Do I have the right to relicense a fork of it under GPL3?

I hate to bother you with these "license woes." If you could offer me any guidance or constructive feedback, I'm all ears. Thanks

vhelin, 2020-06-13:
I have to admit that I don't know. I found this:

http://www.groklaw.net/article.php?story=20060118155841115

but I just cannot spend my time reading through such. :)

Any experts on this subject here? Also: is GPLv3 better for WLA DX than GPLv2? Can we change from using GPLv2 to GPLv3?

vhelin, 2020-06-13:
I read in Wikipedia about Tivoization (GPLv3 doesn't allow it, GPLv2 allows it), and I think it doesn't apply to WLA DX's case. But regarding other changes in the license, would GPLv3 be better for WLA DX?

Drenn1, 2020-07-10:
You don't necessarily need to switch to GPLv3 - you could simply use "GPLv2+", which will allow people to "upgrade" your code to a newer GPL version when they fork it. I think you would just need to add a small clause to the license that "any later version" of the license can be used.

I don't have an opinion on whether GPLv3 is a better license though. It might be best practice to get approval from significant contributors if you want to change the license. Personally I have no objections.

nonchip, 2020-11-10:
> It might be best practice to get approval from significant contributors if you want to change the license.

it's also a legal requirement, since all contributions are licensed under the license at that time. just like someone else can't fork the project and change the license without permission, the project itself can't change the license without the permissions of the authors. unless of course the license at that time (GPLv2) includes a clause explicitly allowing this.

vhelin, 2022-08-29:
As we fixed issue https://github.com/vhelin/wla-dx/issues/515 we are now using GPL-2.0-or-later which means that WLA DX is now definitely GPL3 compatible. Thus I'm closing this issue.

closed 2022-08-29 by vhelin