vhelin/wla-dx #323: array of STRUCT 'spacing' enhancement

sverx, 2020-06-15:
I'm trying to see if there's a way to create an array of STRUCT(s) where each element of the array is evenly spaced from the previous BUT at a different distance than that of its size.

I mean, as an example, say I have a simple DSTRUCT like this one:

```
.STRUCT enemy
id      db
y       db
x       db
.ENDST
```
so if in a **ramsection** I declare this:

`enemies INSTANCEOF enemy 12`

I get a _packed_ array of 12 _enemy_ elements one after another.

Now, I would like to have them 'spaced' in a precise way instead, and the RAM between the elements should still remain available for other ramsection.

I thought probably something along this line would do:

`<instance name> INSTANCEOF <struct name> [optional, the number of structures [optional SPACING <value>]]`

as in 

`enemies INSTANCEOF enemy 12 SPACING 256`

(the spacing value must not be less than the size of the struct of couse!)

Thus: enemies[1] would be at enemies[0]+256, enemies[2] would be at enemies[0]+512 and so on, with the interstitial RAM blocks still available for other ramsections.
(of course the word 'SPACING' might not be the correct one, but PADDING doesn't sound correct to me either...)

Is there an alternative way to do this already (apart from forced placement of each element of the array or by wasting RAM inside the struct with useless padding)?

vhelin, 2020-06-17:
Adding support for SPACING keyword and functionality would be easy, but doing it so that the space skipped would be available for other RAMSECTIONs would be tricky as WLALINK would need to place RAMSECTIONs inside RAMSECTIONs - there is no support for holes inside sections.

One thing that comes into my mind is that could you just create a RAMSECTION for each instance of "enemy" with ALIGN? If doing that by hand is too laborous, could you create a .MACRO that would create those RAMSECTIONs? Would that work?

sverx, 2020-06-17:
'enemy' was just an example, and BTW it seems we can't create RAMSECTIONs using MACROs as they need a unique name that is a string, for instance

Of course there's no support for holes into RAMSECTIONs, what I expected is that it would actually  reserve different blocks (after the first) as if they were additional FORCEd (to a specific address based on the first one's) RAMSECTIONs, if you get what I mean.

as in:
```
.RAMSECTION "enemy 0" ALIGN 256 FREE
  enemies.0 INSTANCEOF enemy
.ENDS

.RAMSECTION "enemy 1" FORCE <at enemy0+256> 
  enemies.1 INSTANCEOF enemy
.ENDS
```
also, I never said it was an easy thing to do, so no problem!

vhelin, 2020-06-17:
I got this to work:

```
.macro makeRAMSECTIONs
.org \@ * 256
.ramsection "enemy \@" force keep
enemies.\@ INSTANCEOF enemy
.ends
.endm

        .rept 10
        makeRAMSECTIONs
        .endr
```

I guess doing the alignment manually using .ORG is not very dynamic... Any ideas?

sverx, 2020-06-17:
I've never seen before this `\@`... is that the same as defining an INDEX for the .REPEAT?

btw yeah, manual alignment isn't very flexible but still a potential workaround as long as there's no better alternative - so thank you!

vhelin, 2020-06-17:
\@ is the number of makeRAMSECTIONs calls in my example, so it's kinda like the INDEX for .REPEAT...

vhelin, 2020-06-17:
I guess you might try feeding INDEX to that .MACRO and then use \1 inside the .MACRO instead of \@... ?

sverx, 2020-06-17:
I seem to remember I had already tried putting RAMSECTIONs inside a .REPEAT (with no success), but I never tried using a .MACRO

Well, I'll try this later when I'll have to tackle this.

vhelin, 2021-05-29:
I'm closing this issue as I think that my makeRAMSECTIONs .MACRO is a working solution to this problem. And instead of giving .ORG to every .RAMSECTION you could make now a chain of .RAMSECTIONs using AFTER and OFFSET (256, 512, ...) keywords and lock only the first .RAMSECTION in place.

If these don't work, feel free to reopen this issue.

closed 2021-05-29 by vhelin