vhelin/wla-dx #325: instanceof structs generates useless labels, element count starting from 1

sverx, 2020-06-22:
If we have this code:

```
.struct enemy
    HP                      db
    posx                    db
    posy                    db
.ends

.ramsection "enemies" slot 3
  enemies instanceof enemy 2
.ends
```
we get all these labels:

```
enemies
enemies.HP
enemies.posx
enemies.posy
enemies.1.HP
enemies.1.posx
enemies.1.posy
enemies.2.HP
enemies.2.posx
enemies.2.posy
```

First question is: aren't `enemies.HP`, `enemies.posx` and `enemies.posy` totally useless, at least when you specify that there's more than one instance of _enemy_?

Second question is: why is the element counter starting from 1? Can we somehow specify that we prefer them to start from 0 instead?

vhelin, 2020-07-09:
> First question is: aren't `enemies.HP`, `enemies.posx` and `enemies.posy` totally useless, at least when you specify that there's more than one instance of _enemy_?

Now that I think of it, they might be useless when there is more than once instance... Removing these labels in this case would perhaps break some old projects, I don't know, but I guess the damage wouldn't be big and easily repairable.

> Second question is: why is the element counter starting from 1? Can we somehow specify that we prefer them to start from 0 instead?

Yeah, well, they are labels, not indices so I thought that 1 would be more appropriate. 

Hmm, perhaps I can make something like

  enemies instanceof enemy 2 startfrom 0

work...

sverx, 2020-07-10:
This is great. I confirm the useless labels are gone when multiple instances of a struct are requested and the **STARTFROM** keyword seems to do its job fine. :heart: 

I'm quite curious anyway about the fact that it seems also to accept negative values. I wouldn't ever use them for sure, but aren't labels with a _minus sign_ in their names invalid?

vhelin, 2020-07-10:
There already is a check for negative values, should not accept them. Also tested with -1 and didn't work.

    if (compare_next_token("STARTFROM") == SUCCEEDED) {
      skip_next_token();

      q = input_number();

      if (q == FAILED)
        return FAILED;
      else if (q == SUCCEEDED) {
        if (d < 0) {
          snprintf(emsg, sizeof(emsg), "STARTFROM needs to be >= 0.\n");
          print_error(emsg, ERROR_DIR);
          return FAILED;
        }
        start_from = d;
      }
      else {
        snprintf(emsg, sizeof(emsg), "STARTFROM needs a number >= 0.\n");
        print_error(emsg, ERROR_DIR);
        return FAILED;
      }
    }

sverx, 2020-07-10:
Uff!!! My build script was wrong and I didn't see the error appear anywhere!

Sorry, now I found it: `DIRECTIVE_ERROR: STARTFROM needs a number >= 0.`

So everythings' good! Thanks! :smiley:

closed 2020-07-10 by sverx