vhelin/wla-dx #326: Unknown symbol

LucianoTheWindowsFan, 2020-06-22:
I wanna play music on my SNES game but the compiler says that there is an unknown symbol so it cannot compile, why I'm having this error?

vhelin, 2020-06-22:
Well, can you post here the line of the source code that is giving WLA problems (and the full error message)?

LucianoTheWindowsFan, 2020-06-22:
Here it is:
` ; SNES Initialization Tutorial code
 ; This code is in the public domain.
 
 .include "Header.inc"
 .include "Snes_Init.asm"
.INCBIN "bibendum.bin"
 
 ; These definitions are needed to satisfy some lines in "Snes_Init.asm".
 .define BG1MoveH $7E1A25
 .define BG1MoveV $7E1A26
 .define BG2MoveH $7E1A27
 .define BG2MoveV $7E1A28
 .define BG3MoveH $7E1A29
 .define BG3MoveV $7E1A2A
 
 ; Needed to satisfy interrupt definition in "Header.inc".
 VBlank:
   rti
 
 .define AUDIO_R0 $2140
 .define AUDIO_R1 $2141
 .define AUDIO_R2 $2142
 .define AUDIO_R3 $2143
 
 .define XY_8BIT $10
 .define A_8BIT  $20
 
 .define musicSourceAddr $00fd
 
 ; The SPC file from which we read our data.
 .define spcFile "test000.spc"
 
 ; The address in SPC RAM where we put our 15-byte startup routine.
 .define spcFreeAddr $ffa0
 
 ; The first half of the saved SPC RAM from the SPC file.
 .bank 1
 .section "musicData1"
 spcMemory1: .incbin spcFile skip $00100 read $8000
 .ends
 
 ; The second half of the saved SPC RAM from the SPC file.
 .bank 2
 .section "musicData2"
 spcMemory2: .incbin spcFile skip $08100 read $8000
 .ends
 
 .bank 0
 .section "MainCode"
 
 ; The rest of the saved SPC state from the SPC file.
 dspData:  .incbin spcFile skip $10100 read $0080
 audioPC:  .incbin spcFile skip $00025 read $0002
 audioA:   .incbin spcFile skip $00027 read $0001
 audioX:   .incbin spcFile skip $00028 read $0001
 audioY:   .incbin spcFile skip $00029 read $0001
 audioPSW: .incbin spcFile skip $0002a read $0001
 audioSP:  .incbin spcFile skip $0002b read $0001
 
 Start:
     ; Initialize the SNES.
     Snes_Init
 
     jsr LoadSPC
 
     ; Set the background color to green.
     sep     #$20        ; Set the A register to 8-bit.
     lda     #%10000000  ; Force VBlank and set brightness to 0%.
     sta     $2100
[... 261 lines cut ...]
     waitForAudio0M
 
     rts
 
 StartSPCExec:
     ; Starting address is in x.
 
     ; Wait until audio0 is 0xbbaa
     sep     #A_8BIT
     lda     #$aa
     waitForAudio0M
 
     ; Send the destination address to AUDIO2.
     stx     AUDIO_R2
 
     ; Send $00cc to AUDIO0 and wait for echo.
     lda     #$00
     sta     AUDIO_R1
     lda     #$cc
     sta     AUDIO_R0
     waitForAudio0M
 
     rts
 
 .ends`

vhelin, 2020-06-22:
Uhh, something must have gone wrong in copying the code here as I don't see

-lda.l

anywhere near line 136... Did you copy and paste the code from somewhere else? If you really have

-lda.l

on that line, put one space after the minus, does that help? I'm assuming it's a local label in this case (the minus)...

LucianoTheWindowsFan, 2020-06-22:
Tried it, didn't work, it seems like it was for an older version, could you convert it to work on newer versions?

vhelin, 2020-06-22:
Put all the files available somewhere and I can take a look at it as a whole... This copy pasted version I cannot convert as it contains too many white space problems I'm sure are not in the original source file.

LucianoTheWindowsFan, 2020-06-22:
I tried it without the SPC700 code, it worked! could you give me a more compatible version of the SPC700 code.¿?

LucianoTheWindowsFan, 2020-06-22:
Tries it win the space, didn't work, tried it without the music code, it worked! what's wrong with it then?

vhelin, 2020-06-22:
I don't see any SPC700 code in this file, only 65816 code... If you can give me a link to the whole project I can take a look at it.

Neui, 2020-06-22:
> Uhh, something must have gone wrong in copying the code here as I don't see
>
> -lda.l
>
> anywhere near line 136... Did you copy and paste the code from somewhere else? If you really have
>
> -lda.l
>
> on [...]

It is (likely) on the line with `-lda.l spcMemory1,x ; Copy byte from first music bank.` since `-` is also used to create an bulled point.

Honestly, this is out of scope for the WLA-DX project, since it is just an assembler (with an linker), not an code/program repository of the architectures it supports.

vhelin, 2020-06-22:
I fixed all white space problems, removed .INCLUDEs and .INCBINs as I don't have them, added a SNES memory map, and added that minus to line 136:

$ ./binaries/wla-65816.exe -o joo.o joo
DISCARD: joo: Discarding an empty section "musicData1".
DISCARD: joo: Discarding an empty section "musicData2".

-> assembles just fine.

Which version of WLA is the original poster using here? I'm using the latest 9.11a in GitHub...

vhelin, 2020-06-22:
> It is (likely) on the line with `-lda.l spcMemory1,x ; Copy byte from first music bank.` since `-` is also used to create an bullet point in MarkDown (so you don't find the `-lda.l` but `lda.l` with an bullet in front of it).
> 
> Honestly, this is out of scope for the WLA-DX project, since it is just an assembler (with an linker), not an code/program repository of the architectures it supports.

I also assume that

- lda.l ...

works and

-lda.l doesn't work...

vhelin, 2020-06-22:
Also make sure that the minus is at the very beginning of that line so it's interpreted as a label. Otherwise this doesn't work, if you have white space on the left side of that minus...

LucianoTheWindowsFan, 2020-06-22:
I used

> I fixed all white space problems, removed .INCLUDEs and .INCBINs as I don't have them, added a SNES memory map, and added that minus to line 136:
>
> $ ./binaries/wla-65816.exe -o joo.o joo
> DISCARD: [...]

I used ver 9.10

vhelin, 2020-06-22:
Here you can download the latest EXEs for Windows:

https://dev.azure.com/villehelin0486/villehelin/_build/results?buildId=163&view=artifacts&type=publishedArtifacts

If that doesn't work, email me that source file (ville dot helin at gmail dot com) and I can take a look at how you are using the whitespace in that file...

LucianoTheWindowsFan, 2020-06-22:
> Here you can download the latest EXEs for Windows:
>
> https://dev.azure.com/villehelin0486/villehelin/_build/results?buildId=163&view=artifacts&type=publishedArtifacts
>
> If that doesn't work, email me [...]

Tried new version, didn't work.

vhelin, 2020-06-22:
Can't help you more unless you start emailing me the files... Or upload them to some file share service and post a link to that here...

LucianoTheWindowsFan, 2020-06-22:
> Can't help you more unless you start emailing me the files... Or upload them to some file share service and post a link to that here...

the download works, i meant the asm didn't work.

LucianoTheWindowsFan, 2020-06-22:
> Put all the files available somewhere and I can take a look at it as a whole... This copy pasted version I cannot convert as it contains too many white space problems I'm sure are not in the original source file.

Here's the file
[code.txt](https://github.com/vhelin/wla-dx/files/4814943/code.txt)

vhelin, 2020-06-22:
None of the labels in that file start at the beginning of the line, they all have one space on the left side of them...

vhelin, 2020-06-22:
So, for example line 137. What you have now is

" -  lda.l spcMemory1,x     ; Copy byte from first music bank."

but for it to work it needs to be

"-  lda.l spcMemory1,x     ; Copy byte from first music bank."

(remove the quotation marks)

vhelin, 2020-06-22:
Actually, now that I think of this, it might be possible to enhance the preprocessor to handle -/--/--- and +/++/+++ local labels when they are not at the beginning of the line?

LucianoTheWindowsFan, 2020-06-22:
> - lda.l spcMemory1,x ; Copy byte from first music bank.

Tried it, still didn't work.

vhelin, 2020-06-22:
Here you can download the new Windows EXEs. It should work now even with the white spacing in your code.txt

https://dev.azure.com/villehelin0486/villehelin/_build/results?buildId=166&view=artifacts&type=publishedArtifacts

LucianoTheWindowsFan, 2020-06-22:
> Here you can download the new Windows EXEs. It works now even with the white spacing in your code.txt
> 
> https://dev.azure.com/villehelin0486/villehelin/_build/results?buildId=166&view=artifacts&type=publishedArtifacts

Tried it, it worked, but when I test it, it black screened and no audio could be heard.

vhelin, 2020-06-22:
That's something I cannot help you with as I didn't write that code...

LucianoTheWindowsFan, 2020-06-22:
Source Code:
[https://en.wikibooks.org/wiki/Super_NES_Programming/Loading_SPC700_programs#Complete_Source_Code](https://en.wikibooks.org/wiki/Super_NES_Programming/Loading_SPC700_programs#Complete_Source_Code)

vhelin, 2020-06-22:
Perhaps you need to ask the person who wrote that code?

vhelin, 2020-06-22:
I haven't written a single line of code for the SNES so I have no idea...

closed 2020-06-23 by vhelin