vhelin/wla-dx #327: section appendto 'offset' enhancement

sverx, 2020-07-01:
Enhancing on #324...
Now that we can place a section at an offset of an aligned address, what about offsetting an appended section too?

For example, we have a LUT that we need to have aligned at 256.
First 16 values are unused, then there are 112 actual data bytes...

```
.SECTION "myLUT" ALIGN 256 OFFSET 16 FREE
.db  $4, $6, $8, $15, $18, $4, $6, $8, $15, $18, $7, $6, $8, $10, $18, $4, $6, $8, etc etc etc...
.ENDS
```
then there are some 64 unused values, then again some actual data. If we could append a new section specifying an offset, that would work with no waste...

To do that, for instance, APPENDTO might accept an OFFSET too, as in:
```
.SECTION "myLUT 2nd part" APPENDTO "myLUT" OFFSET 64
.db $18, $7, $6, $8, $10, $18, $4, $6, $8, etc etc etc...
.ENDS
```
so to have a section appended to another 'with some space in between'.

Wouldn't it be cool? :smile:

vhelin, 2020-07-01:
Currently this would be tricky to do as when we append sections we just merge the datas of the two sections into one bigger section. So either we'd need to add support for holes into sections (might get ugly) or handle APPENDTO sections with OFFSET totally differently (perhaps the better way)...

sverx, 2020-07-01:
Yes, I see your point in simply merging the APPENDed sections, as that's contiguous data anyway.
Here it's trickier because the 'following' section is sort of at a FORCEd address, but that address depends on the address of another section, its length and the offset parameter specified for this very same section, all summed up...

sverx, 2020-07-01:
also, a new approach might also fix #313 (I mean, if the APPENDed section doesn't merge with the other one it will still exist...)

vhelin, 2020-07-01:
Would this be enough?

.SECTION "myLUT 2nd part" FIXTO "myLUT" OFFSET 64
.db $18, $7, $6, $8, $10, $18, $4, $6, $8, etc etc etc...
.ENDS

We we wouldn't merge the datas, "myLUT 2nd part" couldn't make references to "myLUT"'s local labels, but FIXTO would just be a note to the section placer how to place "myLUT 2nd part"?

This would be relatively easy to handle in WLALINK...

sverx, 2020-07-01:
If you think APPENDTO with OFFSET isn't good, this FIXTO does the same anyway (I didn't know APPENDTO would allow to reference local labels in the first section...) so it's up to you!

Also, no rush.

sverx, 2020-07-01:
What about AFTER? You might even not specify any OFFSET and get two sections APPENDED anyway...?

vhelin, 2020-07-01:
Wouldn't "OFFSET 0" be the same as AFTER? I think it's good to keep OFFSET there... And FIXTO OFFSET 0 wouldn't be the same as APPENDTO as FIXTO wouldn't merge the sections...

sverx, 2020-07-01:
Sorry I meant to say AFTER instead of FIXTO as in

`.SECTION "myLUT 2nd part" AFTER "myLUT" OFFSET 64`

or even with no OFFSET as in

`.SECTION "myLUT 2nd part" AFTER "myLUT"`

which would do the same as having OFFSET 0

vhelin, 2020-07-01:
That sounds good!

sverx, 2020-07-01:
Yeah sorry for confusion :smile:

vhelin, 2021-05-01:
It took a couple of days of thinking, planning and coding to get this issue (hopefully) fixed (and #313)... :) It'd bre great if you could test the latest sources.

closed 2021-05-25 by vhelin