vhelin/wla-dx #328: Make "ASCIITABLE" entries work with inline characters

Drenn1, 2020-07-10:
Apparently, asciitable has no effect on things like this:

```
ld a,'A'
```

This will always turn out to be ``ld a,$41`` even if it's been redefined in the ASCIITABLE. I think it would make more sense if the ASCIITABLE affected this.

vhelin, 2020-07-10:
Doing so might have the effect of breaking lots of old code without people seeing error messages or warnings. Perhaps a new directive that'd do this would be a better idea?

Kroc, 2020-07-10:
Oh yes that would break some of my code! I'm thinking that a prefix or suffix would suffice, e.g `lda # ASC 'a'`

Kroc, 2020-07-10:
Ah, another suggestion would be that double quotes get translated, where single quotes don't

```
lda # "a" ; gets translated
lda # 'a' ; does not get translated
```

Drenn1, 2020-07-10:
Kinda curious why ASCII values would be used when the asciitable is not actually ASCII, but you're right - silent breakage is bad. I don't think using double quotes would fix the issue either (and it would be pretty confusing anyway).

So, yeah, perhaps a directive or assembler flag.

vhelin, 2020-07-11:
Hmm, what it you defined .ASCIITABLE with GLOBAL?

    .ASCIITABLE GLOBAL
    MAP "A" TO "Z" = 0
    MAP "!" = 90
    .ENDA

The GLOBAL would mean that all the 'a' etc that are parsed later will go through the .ASCIITABLE remapping? Without GLOBAL it'd work as before...

Kroc, 2020-07-11:
Adding more & more unique keywords makes the assembler harder to learn and more difficult to port code to/from. I think this issue is a case for a command switch, since it should effect all parsing of all files from the get-go. In your .ASCIITABLE example there are many questions that arise like what happens to character strings that occur before an `.ASCIITABLE GLOBAL` definition. What if there are multiple such definitions. What if two ASCIITABLES are global and what if some are and some aren't.... etc. The keyword is not worth the hassle IMO

vhelin, 2020-07-11:
None of the current directives affect the code that was issued before it, and I was thinking that .ASCIITABLE GLOBAL would be no exception - It'd affect only the code that comes after issuing it. If multiple .ASCIITABLE GLOBAL definitions, only the latest would have effect, just like how multiple .ASCIITABLE definitions work.

Would .GLOBALASCIITABLE that'd do the same be better?

Kroc, 2020-07-13:
Most of the choices break the law of least-surprise; you can't, just by looking at a file in isolation, tell which byte is going to be written. Using a prefix/suffix keyword for characters would at least be a localised effect that keeps questions of scope spiralling out of control. Perhaps consider a more complex system for v10; e.g. allow the naming of ASCII tables so that multiple ones can be in play and they don't need to be re-included each time you want to swap back to them.

vhelin, 2020-07-13:
Would you like to try to create the specification of this feature? I'm happy just to code it, and I'd like to code something that's useful for all. If I make the specs here there is the danger that I'll design something that sucks as I have trouble seeing the big picture in this case. :)

Drenn1, 2020-07-13:
I sympathize here, most solutions to this that I can think of suck =)

If you want to avoid ambiguity, using a sort of "built-in" function might be a decent way to handle it, ie:

```
ld a, ASCIIMAP('A')
```

I have in mind that "ASCIIMAP" would be a built-in function which works in the same way as assembly-time math macros suggested in #309, the only difference being it would be a pre-defined function.

vhelin, 2021-03-14:
How about this? I called the function ASC() as it does what .ASC does...

Kroc, 2021-03-16:
Nice and clear!

Drenn1, 2021-03-16:
I really do wish that we didn't need to wrap it in a function, but if that's not acceptable for compatibility reasons (& other reasons discussed above), then I suppose this is fine.

closed 2021-04-27 by vhelin