vhelin/wla-dx #331: Continuous builds, especially for making releases

maxim-zhao, 2020-08-17:
It's not too hard to set up CI for WLA DX as it builds cleanly with CMake. This could be used to produce per-commit builds for people and this would save Ville having to get his hands dirty to make Windows builds.

Here's the [Appveyor build](https://ci.appveyor.com/project/maxim-zhao/wla-dx-6yqhq) I set up a long time ago, but Github Actions can work too:

```
version: 1.0.{build}
image:
- Visual Studio 2019
- Ubuntu
install:
- cmd: pip install -U sphinx
build_script:
- cmd: |-
    cmake .
    cmake --build . --config Release -j
    7z a wladx.zip %APPVEYOR_BUILD_FOLDER%\binaries\Release\*.exe
    7z a wladx.zip %APPVEYOR_BUILD_FOLDER%\doc\gen-man -r
    7z a wladx.zip %APPVEYOR_BUILD_FOLDER%\doc\gen-shtml -r
    7z a wladx.zip %APPVEYOR_BUILD_FOLDER%\doc\gen-stext\*.txt
artifacts:
- path: '*.zip'
```

vhelin, 2020-08-18:
Cool, thanks!

We already though have Azure CI for Linux and Windows: https://github.com/vhelin/wla-dx (scroll down and in README.md you'll see two Azure icons, click them and you can access per commit artefacts)...

The problem we have currently is that our Windows binaries are (I guess your Appveyor builds are the same) are built so that the executables work only on Windows 10. What I'm hosting on my personal server are Windows 7 compatible executables, built using some old (2012?) Visual Studio on a Windows 7 PC (my parents' PC)... It's not a big hassle to do the builds on that computer, it's just that I rarely have access to it. And I guess Azure doesn't support Windows 7 VM's... ?

vhelin, 2020-08-18:
Can you select Visual Studio 2012 in AppVeyor?

Should we add link to that AppVeyor page to README.md as well?

maxim-zhao, 2020-08-18:
I hadn't figured out how to get the exes from the Azure build, it is a bit tricksy :) and the zip is missing docs and licences...

As it's not my repo, the appveyor builds aren't continuous so they're a bit useless. Visual Studio can target older Windows - although I don't have any to test on - including XP up to VS 2017. I thought latest VS2019 builds are fine for Win7 provided the CRT is installed.

I dug a bit deeper and realised my build script is here: https://gist.github.com/maxim-zhao/e92786ef82ff5949441bdf4a5ca9a6d6 as it needs to differ between Ubuntu and Windows. I added in other VS versions to see what difference it makes, but it only goes back to 2013 - see https://ci.appveyor.com/project/maxim-zhao/wla-dx-6yqhq/builds/34728682 . I also tried targeting Windows XP, which seemed to succeed - https://ci.appveyor.com/project/maxim-zhao/wla-dx-6yqhq/build/job/45hg83rc5gdmbam3 although may also be 64-bit.

vhelin, 2020-08-19:
Does anyone here have a Win7 or XP machine and could test if maxim-zhao's builds work?

vhelin, 2020-08-19:
And if anyone here know how to turn this

  https://github.com/vhelin/wla-dx/blob/master/azure-pipelines-windows.yml

into something that builds Windows 7 compatible (or even Windows XP) executables, please let us know!

vhelin, 2020-08-19:
Or should the .vcxproj files inside these directories modified somehow? I just don't know. :)

  https://github.com/vhelin/wla-dx/tree/master/windows

StarWolf3000, 2020-08-24:
You can also set up Actions or external CI to build using MSYS2 mingw-w64. I'm always building my local repo copies with mingw-w64 i686 and x86-64 from MSYS2 project. This way they're not limited to Windows 10 execution environment, but require at least Vista, as XP support has been dropped at least 2 years ago in mingw-w64.
I have stopped updating my environment after GCC 9.3, as the MSYS2 team has changed package format for everything, making updates a very difficult task, but this does not affect fresh installs like CI would do.

vhelin, 2020-08-24:
But if you use MSYS2 to build executables, don't they need MSYS2 to run?

vhelin, 2020-08-24:
Anyway, I was able to configure the Azure Pipelines I'm using to produce Windows XP compatible executables:

https://stackoverflow.com/questions/49516896/how-to-install-build-tools-for-v141-xp-for-vc-2017

Changed the platform tools from "v141" to "v141_xp" in the project files, and set Azure to use vmImage "vs2017-win2016" instead of "windows-latest" / "windows-2019"...

I also installed v141_xp locally on my laptop so I can make XP compatible exes also any time I wish.

Everybody can now get the latest sources built for Windows XP from the README.md's link to Azure Pipelines:

https://dev.azure.com/villehelin0486/villehelin/_build/latest?definitionId=2&branchName=master

(click on "1 published")

closed 2020-08-24 by vhelin

cr1901, 2020-08-24:
>But if you use MSYS2 to build executables, don't they need MSYS2 to run?

_It depends._ MSYS2 comes with three shells- `MSYS2`, `MINGW32`, and `MINGW64`. The latter two create executables that do not require the Cygwin dll, but _may_ require you to bundle `libgcc` with the executables, since it's unlikely to be present unless you already have gcc installed (unless there's a way to get CMake to do this for you).

StarWolf3000, 2020-08-24:
By default, you dynamic link against the default Windows runtime msvcrt.dll and static link libgcc, when using MINGW32/MINGW64 environment.