vhelin/wla-dx #333: Invalid default value in SMS header

HerrSchatten, 2020-09-08:
If not explicitly specified in SMSHEADER, the upper nibble of the byte at $7FFF defaults to 0, which is an invalid value for region information. 
In older versions of WLA-DX the default value used to be 4 (= export SMS), which it should be, as it's the only value any of the BIOSes actually checks for.

sverx, 2020-09-10:
https://wla-dx.readthedocs.io/en/latest/asmdiv.html#smsheader

REGIONCODE should default to 4 when unspecified and ROMSIZE to:

- $a (if ROM size is 8 KB, SMSHEADER goes at $1FF0)
- $b (if ROM size is 16 KB, SMSHEADER goes at $3FF0)
- $c (if ROM size is 32 KB or more, SMSHEADER goes at $7FF0)

because other values are a bad idea and might even trigger the BIOS check bug (see: https://www.smspower.org/Development/ROMHeader#ROMSize0x7fff05Bytes )

vhelin, 2020-09-18:
Hi! Sorry for the delay, I've been really busy with other things... I hope it's fixed now, thanks for the report!

HerrSchatten, 2020-09-19:
The region code is fixed now, thanks.

However, romsize only defaults to the correct value for the automatically calculated checksum if smsheader is present (even if there are no values set within smsheader), otherwise it defaults to zero, creating a mismatch between the romsize nibble in the header and the automatically calculated checksum. See sverx's post above for the right default values.

vhelin, 2020-09-19:
Ah, the ROMSIZE value needs to be calculated and written even only .SMSCHECKSUM is defined? I missed that in a hurry. :) Does it work now better?

Kroc, 2020-09-19:
This has not fixed the checksum being wrong in Sonic1-Z80-ASM, despite no other bytes in the ROM (including the other header properties) being different;

[image]

Original ROM on top, my build on the bottom. Only these two bytes differ.

Kroc, 2020-09-19:
Fixed for me, thanks so much!

vhelin, 2020-09-19:
I'll add a test for this so if it breaks again in the future we can see that immediately. :)

HerrSchatten, 2020-09-20:
Works perfectly now, thanks.

closed 2020-09-20 by HerrSchatten