vhelin/wla-dx #335: scp700 - bbs and bbc instructions not printed properly in list file

everything8215, 2020-10-08:
When I generate a list file for code assembled with wla-spc700, the output bytes are out of alignment whenever there is a bbs or bbc instruction. These should be 3-byte instructions, but it only prints 1 byte. The remaining 2 bytes get pushed down to the next line. For example, if I assemble

`    bbc     $bf.3,@_02f1
    jmp     !main`

I get

`$F0                                         bbc     $bf.3,@_02f1
$73 $BF                                     jmp     !main
$5F $8C $02                             `

But it should be

`$F0 $73 $BF                                 bbc     $bf.3,@_02f1
$5F $8C $02                                 jmp     !main`

I think there might be an issue with opcode types c and d in decode_spc700.c, but I couldn't figure out how it determines the number of bytes to print on each line in the list file.

Note that this only affects the list file. The output file still gets assembled properly.

vhelin, 2021-03-07:
Thanks for the report, this issue was on multiple targets. I hope it's fixed now...

everything8215, 2021-03-12:
Yep, it's fixed. Thanks!

closed 2021-03-12 by everything8215