vhelin/wla-dx #336: interpolation issues with REPEAT / MACRO

nonchip, 2020-11-09:
i'm trying to generate reset vectors according to the following pseudo-code:
```
for \R from $00 to $38 step $08 do
  .ORGA \R
  .SECTION "BOOT_RESET_\R" FORCE KEEP
    boot_rst_\R:
      jp ($F000 + \R)
  .ENDS
done
```

what I've tried:
```asm
.REPEAT 8 INDEX IDX
  .ORGA (IDX * $0008)
  .SECTION "BOOT_RESET_IDX" SIZE $0008 FORCE RETURNORG KEEP ; <-- errors
    boot_rst_IDX:
      jp ($F000+ IDX)
  .ENDS
.ENDR
```
errors out with:
> DIRECTIVE_ERROR: SECTION "BOOT_RESET_IDX" was defined for the second time.
since there's no string interpolation happening

so i thought i'd be smart and use a macro instead because those have interpolation:
```asm
.MACRO make_rst
  .ORGA (\1 * $0008)
  .SECTION "BOOT_RESET_\1" SIZE $0008 FORCE RETURNORG KEEP ; <-- errors
    boot_rst_\1:
      jp ($F000+ \1)
  .ENDS
.ENDM
.REPEAT 8 INDEX IDX
make_rst IDX
.ENDR
```
guess what, still errors out with the SAME error, somehow expanding the macro 8 times but replacing `\1` with the string "IDX" instead of the index.

vhelin, 2021-02-21:
The "BOOT_RESET_\1" should have expanded to "BOOT_RESET_0", "BOOT_RESET_1", etc., but the string processor just stupidly copied the plain text of the argument to \1... I have no idea why we didn't implement it properly as the string processor has all the information it needs to do it nicely.

Anyway, just fixed it. :) If you now try the latest sources your .MACRO should work as you wanted...

PS. Sorry for the huge delay, I totally missed this issue and only this weekend had lots of free time to go through the pending issues list...

closed 2021-05-25 by vhelin

nonchip, 2021-05-30:
afraid i switched my entire codebase over to `sdcc` in the meantime (who then managed to mess up `IN/OUT` instructions, FUN!) so i can't really test it, also i doubt i'll be wanting to switch back to pure assembler for this project anyway.

vhelin, 2021-05-30:
No problem, I'm pretty confident I got this one fixed. :)