vhelin/wla-dx #337: Linking RAMSECTIONS with libraries

Scryfox, 2020-12-01:
Hello! I am trying to link a ramsection that is in a library, but I am getting an error <code>wla-65816 -o main.o main.asm
wla-65816 -l ./libraries/controllers.lib ./libraries/controllers.asm
wlalink -v Linkfile main.sfc
./libraries/controllers.lib: ./libraries/controllers.asm: CHECK_RAMSECTIONS: RAM section "Controller Status" has no BANK/SLOT. Give them in the linkfile under [ramsections].</code>

However, in my link file (below), I have placed the ramsection with a bank/slot as the documentation shows. If you could provide any guidance, I would really appreciate it! 

[objects]
main.o
[libraries]
bank 0 slot 2 ./libraries/controllers.lib
[header]
[footer]
[ramsections]
bank 0 slot 1 "Controller Status"
[sections]
[definitions]

Scryfox, 2020-12-01:
I am dumb... I forgot to remove the reference to the file in main after moving it into a library... I really appreciate your hard work on this project though, it is really amazing!

closed 2020-12-01 by Scryfox

vhelin, 2020-12-03:
Hi! Good thing you found a solution to the problem. :) Sorry for the late reply, also... :P

I'm happy if you users find WLA DX useful! All improvement ideas are welcome.